x-wrt/x-wrt #1: rt2x00 2.4 radio driver problem

Mighcavots4, 2018-02-01:
It seems something go lost in the move from lede back to openwrt.
Your lede fork was very stable on Mir3.
You openwrt fork has a regression to this old bug.

ptpt52, 2018-02-02:
I have merge the changes from https://github.com/psyborg55/openwrt/commits/master
for rt2x00

Are you sure this will cause problems?

Mighcavots4, 2018-02-05:
Not sure if that patch cause problems, but it does not fix the problem.

On the LEDE fork, I suspect it might be a back ported driver from 
"Backport generated by backports.git v4.14-rc2-1-31-g86cf0e5d"
Those drivers are used in your LEDE fork, but not in the Openwrt one.
Something still crash under heavy load, but the network card does not stay in zombie state.
Most devices automatically reconnect and you don't even notice it.
It looks like this:
"WARNING: CPU: 0 PID: 1559 at net/sched/sch_hfsc.c:1400 0x875edd74 [sch_hfsc@875ec000+0x3360]"

Mighcavots4, 2018-03-03:
Back to this issue.
Yes, it defiantly 2.4G related, the 5G works great and also the Ethernet.
With the latest build, the 2.4G also stops it's being on and not working ( it's back to how it was, with the  LEDE based fork )
Now, it's just all that dumps and warning in the kernel log.
```
11475.476595] WARNING: CPU: 0 PID: 3 at net/sched/sch_hfsc.c:1400 0x875d5d74 [sch_hfsc@875d4000+0x3360]
[11475.486045] Modules linked in: rt2800soc rt2800pci rt2800mmio rt2800lib pppoe ppp_async rt2x00soc rt2x00pci rt2x00mmio rt2x00lib pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink iptable_mangle iptable_filter ipt_ECN ip_tables crc_itu_t crc_ccitt compat act_connmark nf_conntrack act_skbedit act_mirred
[11475.558296]  em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_oneshot ledtrig_heartbeat ledtrig_gpio xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun eeprom_93cx6 leds_gpio ohci_platform ohci_hcd softdog ledtrig_transient ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[11475.620835] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G        W       4.9.82 #0
[11475.628307] Stack : 8042753a 00000043 00000000 00000001 00000000 00000000 00000000 00000000
[11475.636863]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[11475.645428]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[11475.653991]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[11475.662553]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[11475.671115]         ...
[11475.673611] Call Trace:
[11475.675966] [<8000df3c>] 0x8000df3c
[11475.679545] [<8000df3c>] 0x8000df3c
[11475.683104] [<800234a0>] 0x800234a0
[11475.686707] [<875d5d74>] 0x875d5d74 [sch_hfsc@875d4000+0x3360]
[11475.692697] [<80023568>] 0x80023568
[11475.696293] [<875d5d74>] 0x875d5d74 [sch_hfsc@875d4000+0x3360]
[11475.702276] [<8028b690>] 0x8028b690
[11475.705835] [<800573ac>] 0x800573ac
[11475.709417] [<86cb049c>] 0x86cb049c [rt2x00lib@86cb0000+0x7a50]
[11475.715460] [<86cd802c>] 0x86cd802c [rt2800soc@86cd8000+0x970]
[11475.721430] [<802672ec>] 0x802672ec
[11475.724990] [<80025fec>] 0x80025fec
[11475.728569] [<800434cc>] 0x800434cc
[11475.732182] [<8002605c>] 0x8002605c
[11475.735745] [<8003ec50>] 0x8003ec50
[11475.739319] [<8003eda4>] 0x8003eda4
[11475.742900] [<8003ec50>] 0x8003ec50
[11475.746457] [<80356e3c>] 0x80356e3c
[11475.750069] [<8003ec50>] 0x8003ec50
[11475.753628] [<8003b8dc>] 0x8003b8dc
[11475.757284] [<8003b804>] 0x8003b804
[11475.760876] [<80009278>] 0x80009278
[11475.764438] 
[11475.765958] ---[ end trace acc1e4c71dd06d0b ]---
[12698.818172] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.827670] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.837136] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.846596] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.856051] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.865504] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.874954] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.884404] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.893844] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.903294] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.912743] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.922168] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.931613] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.941064] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.950519] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.959970] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.969418] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[12698.978864] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[... cut ...]
[20281.067937] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[20281.077391] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[20281.086844] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[20281.096299] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[20281.105752] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
```

ptpt52, 2018-03-11:
ok, I am fixing this problem.
...good news is:
It is expected to be resolved tomorrow (or maybe after a few hours) and
I will submit the latest firmware to:
https://router-sh.ptpt52.com/rom/

Mighcavots4, 2018-03-22:
Thanks for your work.

I have been testing the latest tree for a few days.
No kernel crash any more, but ... 
Still plenty of : ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
Until it's resolved or my new Xiaomi router 3G arrive,  I plunged to the router a USB WiFi 2.4G adapter that I had around and the 2.4 G works, much better then the original, anyway.

daiaji, 2018-06-11:
@ptpt52 我用K2的话 是不是该试试rt2x00分支？
K2 2.4G确实有问题

ptpt52, 2018-06-11:
@daiaji 这个，主分支就是最新的修复代码。其他分支一般用于给op提交代码用的。

daiaji, 2018-06-11:
@ptpt52 Ok我编译试试
K2的2.4G 一直没法没法连接 然后就用了很长一段时间的官方固件

daiaji, 2018-06-11:
@ptpt52 
系统日志报了这个
```
Sun Jun 10 08:00:36 2018 kern.warn kernel: [  752.950227] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:36 2018 kern.warn kernel: [  752.985706] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:36 2018 kern.warn kernel: [  753.019021] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:37 2018 kern.warn kernel: [  753.158187] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:37 2018 kern.warn kernel: [  753.275878] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:37 2018 kern.warn kernel: [  753.309353] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:37 2018 kern.warn kernel: [  753.418429] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:37 2018 kern.warn kernel: [  753.588381] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:37 2018 kern.warn kernel: [  753.748369] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:38 2018 kern.warn kernel: [  753.857102] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
Sun Jun 10 08:00:38 2018 kern.warn kernel: [  753.891741] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
```
内核日志报了这个
```
[  280.487890] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  280.586373] br-lan: port 3(wlan1) entered blocking state
[  280.597060] br-lan: port 3(wlan1) entered disabled state
[  280.608015] device wlan1 entered promiscuous mode
[  280.617520] br-lan: port 3(wlan1) entered blocking state
[  280.628122] br-lan: port 3(wlan1) entered forwarding state
[  280.737638] br-lan: port 3(wlan1) entered disabled state
[  281.535150] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  281.555322] br-lan: port 3(wlan1) entered blocking state
[  281.565956] br-lan: port 3(wlan1) entered forwarding state
[  297.536229] br-lan: port 3(wlan1) entered disabled state
[  297.546982] br-lan: port 2(wlan0) entered disabled state
[  297.557622] br-lan: port 1(eth0.1) entered disabled state
[  297.595921] device eth0.1 left promiscuous mode
[  297.605250] br-lan: port 1(eth0.1) entered disabled state
[  297.621571] device wlan0 left promiscuous mode
[  297.630595] br-lan: port 2(wlan0) entered disabled state
[  297.664014] device wlan1 left promiscuous mode
[  297.673041] br-lan: port 3(wlan1) entered disabled state
[  297.709784] br-lan: port 1(eth0.1) entered blocking state
[  297.720617] br-lan: port 1(eth0.1) entered disabled state
[  297.731737] device eth0.1 entered promiscuous mode
[  297.779396] br-lan: port 2(wlan0) entered blocking state
[  297.790047] br-lan: port 2(wlan0) entered disabled state
[  297.801009] device wlan0 entered promiscuous mode
[  297.873109] br-lan: port 3(wlan1) entered blocking state
[  297.883789] br-lan: port 3(wlan1) entered disabled state
[  297.951090] device wlan1 entered promiscuous mode
[  298.008326] br-lan: port 3(wlan1) entered blocking state
[  298.018971] br-lan: port 3(wlan1) entered forwarding state
[  298.030001] br-lan: port 2(wlan0) entered blocking state
[  298.040598] br-lan: port 2(wlan0) entered forwarding state
[  298.051574] br-lan: port 1(eth0.1) entered blocking state
[  298.062340] br-lan: port 1(eth0.1) entered forwarding state
[  554.267903] ieee80211 phy1: rt2800mmio_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced reset
[  639.592052] ieee80211 phy1: rt2800mmio_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced reset
[  643.581016] ieee80211 phy1: rt2800mmio_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced reset
[  679.367005] ieee80211 phy1: rt2800mmio_watchdog_tx_dma: Warning - TX queue 0 DMA timed out, invoke forced reset
[  751.552074] device wlan1 left promiscuous mode
[  751.561094] br-lan: port 3(wlan1) entered disabled state
[  751.833158] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  752.885206] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  752.913979] br-lan: port 3(wlan1) entered blocking state
[  752.924660] br-lan: port 3(wlan1) entered disabled state
[  752.935596] device wlan1 entered promiscuous mode
[  752.950227] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  752.985706] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  753.019021] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  753.158187] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  753.275878] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  753.309353] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[  753.418429] ieee80211 phy1: rt2800_config_channel: Warning - Using incomplete support for external PA
[... cut ...]
[  928.795510] br-lan: port 2(wlan0) entered disabled state
[ 1029.065344] br-lan: port 2(wlan0) entered blocking state
[ 1029.075998] br-lan: port 2(wlan0) entered disabled state
[ 1029.086957] device wlan0 entered promiscuous mode
[ 1029.920584] br-lan: port 2(wlan0) entered blocking state
[ 1029.931214] br-lan: port 2(wlan0) entered forwarding state
[ 1098.510681] ieee80211 phy1: rt2800mmio_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced reset
```

可以说很奇怪了
我没有双天线MU-MIMO的手机啊

daiaji, 2018-06-11:
@ptpt52
系统日志
```
Sun Jun 10 12:23:48 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:23:48 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:23:48 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:23:48 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:23:58 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:24:54 2018 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 44:80:eb:cb:e5:88
Sun Jun 10 12:25:55 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:25:55 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:25:55 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:25:55 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:25:58 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:26:16 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:26:16 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:26:16 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:26:16 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:26:19 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:26:52 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:26:52 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:26:52 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:26:52 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:26:57 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:06 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:27:06 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:27:07 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:07 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:27:13 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:44 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:27:44 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:27:44 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:44 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:27:49 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:50 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:27:50 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 2)
Sun Jun 10 12:27:50 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:50 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:27:55 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:27:55 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:27:55 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:27:55 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:28:19 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:29:04 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:29:04 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 2)
Sun Jun 10 12:29:04 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:29:04 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:30:23 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:30:23 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 1)
Sun Jun 10 12:30:23 2018 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:30:23 2018 daemon.info hostapd: wlan1: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
Sun Jun 10 12:30:45 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:30:51 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:30:51 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 2)
Sun Jun 10 12:30:52 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:30:52 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)
[... cut ...]
Sun Jun 10 12:49:17 2018 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:49:21 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: authenticated
Sun Jun 10 12:49:21 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 IEEE 802.11: associated (aid 2)
Sun Jun 10 12:49:21 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 44:80:eb:cb:e5:88
Sun Jun 10 12:49:21 2018 daemon.info hostapd: wlan0: STA 44:80:eb:cb:e5:88 WPA: pairwise key handshake completed (RSN)

```
依旧存在断流的问题
只是没那么经常

ptpt52, 2018-06-11:
@daiaji 
之前是出现2.4G彻底断网，现在还出现彻底断网的情况玛？

daiaji, 2018-06-12:
@ptpt52
wifi断开连接是挺常见的事 但是有时候疯狂连不上 连上就断开 连上就断开
5g也一样
断流是用k2的官改驱动也这样

ptpt52, 2018-06-12:
我是问。用我的固件或者代码，是否解决了 断网的问题。断流还是有的。但是网络不会断。

daiaji, 2018-06-12:
@ptpt52 没解决 很明显的感觉就是 有时候直接连不上 或者连上几秒钟就会断开

daiaji, 2018-09-19:
@ptpt52 我昨晚上构建了你的openwrt-openwrt主干代码
2.4Ghz OK了……
我把2.4Ghz弄成开放网络搞了一个晚上加半天
现在至少还是能连上
虽然内核还是报
```
[ 3381.400021] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.418425] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.436840] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.455249] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.473652] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.492056] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.510460] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.528864] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.547273] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.917752] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.936192] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.954609] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.973017] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3381.991418] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.009815] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.028216] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.046620] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.065018] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.083420] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.101817] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.120213] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.138610] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.157007] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.175408] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.193812] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.212209] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.230606] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.249012] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.267410] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 3382.285811] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
```

ptpt52, 2019-03-06:
fixed in latest code

closed 2019-03-06 by ptpt52