x-wrt/x-wrt #2: WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out

droserasprout, 2018-02-14:
Latest R3 build seems to be completely broken. WAN interface drops packages heavily, router restarts every 10-15 minutes. Not sure if it's upstream bug or your build's one, can't find related bugs on track.

```
root@NATCAP:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 78:11:DC:12:2D:0A  
          inet6 addr: fe80::7a11:dcff:fe12:2d0a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:230519 errors:0 dropped:4 overruns:0 frame:0
          TX packets:330078 errors:7 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:214995099 (205.0 MiB)  TX bytes:366539912 (349.5 MiB)
          Interrupt:5 

root@NATCAP:~# ifconfig eth0.2
eth0.2    Link encap:Ethernet  HWaddr 78:11:DC:12:2D:0A  
          inet addr:176.15.23.109  Bcast:176.15.31.255  Mask:255.255.240.0
          inet6 addr: fe80::7a11:dcff:fe12:2d0a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:198769 errors:0 dropped:155 overruns:0 frame:0
          TX packets:325730 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:209250183 (199.5 MiB)  TX bytes:363788831 (346.9 MiB)
root@NATCAP:~# dmesg
[    0.000000] Linux version 4.9.77 (terminator@SC-GAME) (gcc version 6.3.0 (OpenWrt GCC 6.3.0 r6037-b4f1449) ) #0 Mon Feb 12 00:50:26 2018
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is Xiaomi Mi Router R3
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
debug2: channel 0: window 998953 sent adjust 49623
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 803efde4, node_mem_map 81000020
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=0004027a
[    0.000000] Readback ErrCtl register=0004027a
[    0.000000] Memory: 125396K/131072K available (3435K kernel code, 161K rwdata, 440K rodata, 180K init, 220K bss, 5676K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:256
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: enable autosleep mode
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
[    0.000012] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
[    0.007578] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.070078] pid_max: default: 32768 minimum: 301
[    0.074690] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.081062] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[... 295 lines cut ...]
[  106.212445] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[  106.218617] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c5d000, max=128, ctx=97, dtx=97, fdx=48, next=97
[  106.229206] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c3b000, max=128, calc=13, drx=14
[  108.596675] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  108.603022] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[  108.609209] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c3b000, max=128, ctx=70, dtx=70, fdx=31, next=70
[  108.619794] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c5d000, max=128, calc=114, drx=115
[  111.039740] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  111.046073] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[  111.052244] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c5d000, max=128, ctx=45, dtx=45, fdx=32, next=45
[  111.062833] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c3b000, max=128, calc=108, drx=109
[  140.327210] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  140.333546] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[  140.339714] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c3b000, max=128, ctx=55, dtx=55, fdx=6, next=55
[  140.350209] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c5d000, max=128, calc=18, drx=19
[  144.560456] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  144.566793] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[  144.572960] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c5d000, max=128, ctx=97, dtx=97, fdx=32, next=97
[  144.583544] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c3b000, max=128, calc=90, drx=91
[  148.800531] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[  148.806878] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[  148.813048] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c3b000, max=128, ctx=19, dtx=19, fdx=16, next=19
[  148.823631] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c5d000, max=128, calc=101, drx=102

```

droserasprout, 2018-02-14:
[This thread](https://lists.infradead.org/pipermail/lede-dev/2017-July/008368.html) seems to be pretty related.

ptpt52, 2018-02-15:
ok, I am trying to figure out it

ptpt52, 2018-02-15:
it may be the upstream bug, I recently updated the code with upstream.

droserasprout, 2018-02-15:
I have built clean OpenWrt SNAPSHOT r5671+5-c5ca1c9ab6 build w/ your R3 pull-request merged. Still has some dropped packages on eth0.2, put dmesg is clean and overall performance is fine, also no reboots.

droserasprout, 2018-02-25:
@ptpt52 Last build (20180225) still unusable. Got this panic after the second second of speedtest:
```
[   86.044943] ------------[ cut here ]------------
[   86.049686] WARNING: CPU: 0 PID: 3 at net/sched/sch_generic.c:316 0x8028c23c
[   86.056931] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[   86.064033] Modules linked in: rt2800soc rt2800pci rt2800mmio rt2800lib pppoe ppp_async l2tp_ppp rt2x00soc rt2x00pci rt2x00mmio rt2x00lib pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4
[   86.137276]  nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan iptable_mangle iptable_filter ipt_ECN ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 crc_itu_t crc_ccitt compat br_netfilter natcap fuse sch_cake act_connmark act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_rt ip6t_frag ip6t_hbh ip6t_eui64 ip6t_mh ip6t_ah ip6t_ipv6header ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 eeprom_93cx6 sha256_generic sha1_generic seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ecb ctr cmac uas mmc_block usb_storage mmc_core leds_gpio ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common aead crypto_null cryptomgr crc32c_generic crypto_hash
[   86.299985] CPU: 0 PID: 3 Comm: ksoftirqd/0 Not tainted 4.9.82 #0
[   86.306212] Stack : 8042753a 00000035 00000000 00000001 00000000 00000000 00000000 00000000
[   86.314775]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   86.323320]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   86.331879]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   86.340442]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   86.349005]         ...
[   86.351500] Call Trace:
[   86.353827] [<8000dd68>] 0x8000dd68
[   86.357403] [<8000dd68>] 0x8000dd68
[   86.360961] [<80022bf4>] 0x80022bf4
[   86.364533] [<8028c23c>] 0x8028c23c
[   86.368092] [<80022c2c>] 0x80022c2c
[   86.371730] [<802434e4>] 0x802434e4
[   86.375366] [<8028c23c>] 0x8028c23c
[   86.378925] [<80035aac>] 0x80035aac
[   86.382482] [<8028c0c8>] 0x8028c0c8
[   86.386057] [<8005647c>] 0x8005647c
[   86.389674] [<800566c4>] 0x800566c4
[   86.393234] [<80054b54>] 0x80054b54
[   86.396808] [<80054b60>] 0x80054b60
[   86.400369] [<8002562c>] 0x8002562c
[   86.403925] [<800429ac>] 0x800429ac
[   86.407502] [<80025780>] 0x80025780
[   86.411061] [<8003e200>] 0x8003e200
[   86.414635] [<8003e378>] 0x8003e378
[   86.418193] [<8003e200>] 0x8003e200
[   86.421748] [<80356fdc>] 0x80356fdc
[   86.425321] [<8003e200>] 0x8003e200
[   86.428879] [<8003aec8>] 0x8003aec8
[   86.432440] [<8003adec>] 0x8003adec
[   86.436013] [<80009078>] 0x80009078
[   86.439576] 
[   86.441100] ---[ end trace 0150fd1a0ee4c3a7 ]---
[   86.445833] mtk_soc_eth 10100000.ethernet eth0: transmit timed out
[   86.452147] mtk_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[   86.458298] mtk_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06c48000, max=128, ctx=64, dtx=64, fdx=116, next=64
[   86.468974] mtk_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06c4a000, max=128, calc=112, drx=119
```
Maybe it's reasonable to temporary disable bulding this target?

ptpt52, 2018-02-27:
I notice that this commit may fix the problem
https://github.com/openwrt/openwrt/commit/e9b58b22c100b29c8643e73127f580d24f4ed482

Mighcavots4, 2018-03-02:
I installed your latest tree and it includes the above commit, but still...:

```
 5190.827328] WARNING: CPU: 0 PID: 3 at net/sched/sch_hfsc.c:1400 0x875d5d74 [sch_hfsc@875d4000+0x3360]
[ 5190.836786] Modules linked in: rt2800soc rt2800pci rt2800mmio rt2800lib pppoe ppp_async rt2x00soc rt2x00pci rt2x00mmio rt2x00lib pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink iptable_mangle iptable_filter ipt_ECN ip_tables crc_itu_t crc_ccitt compat act_connmark nf_conntrack act_skbedit act_mirred
[ 5190.909013]  em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_oneshot ledtrig_heartbeat ledtrig_gpio xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun eeprom_93cx6 leds_gpio ohci_platform ohci_hcd softdog ledtrig_transient ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[ 5190.971574] CPU: 0 PID: 3 Comm: ksoftirqd/0 Tainted: G        W       4.9.82 #0
[ 5190.979028] Stack : 8042753a 00000043 00000000 00000001 00000000 00000000 00000000 00000000
[ 5190.987593]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5190.996156]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5191.004718]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5191.013282]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5191.021846]         ...
[ 5191.024347] Call Trace:
[ 5191.026703] [<8000df3c>] 0x8000df3c
[ 5191.030264] [<8000df3c>] 0x8000df3c
[ 5191.033843] [<800234a0>] 0x800234a0
[ 5191.037431] [<875d5d74>] 0x875d5d74 [sch_hfsc@875d4000+0x3360]
[ 5191.043413] [<80023568>] 0x80023568
[ 5191.047010] [<875d5d74>] 0x875d5d74 [sch_hfsc@875d4000+0x3360]
[ 5191.052979] [<8028afd8>] 0x8028afd8
[ 5191.056554] [<8028b690>] 0x8028b690
[ 5191.060115] [<800573dc>] 0x800573dc
[ 5191.063697] [<86cb049c>] 0x86cb049c [rt2x00lib@86cb0000+0x7a50]
[ 5191.069740] [<86cd802c>] 0x86cd802c [rt2800soc@86cd8000+0x970]
[ 5191.075708] [<802672ec>] 0x802672ec
[ 5191.079268] [<80025fec>] 0x80025fec
[ 5191.082842] [<800434cc>] 0x800434cc
[ 5191.086455] [<8002605c>] 0x8002605c
[ 5191.090016] [<8003ec50>] 0x8003ec50
[ 5191.093592] [<8003eda4>] 0x8003eda4
[ 5191.097172] [<8003ec50>] 0x8003ec50
[ 5191.100732] [<80356e3c>] 0x80356e3c
[ 5191.104345] [<8003ec50>] 0x8003ec50
[ 5191.107905] [<8003b8dc>] 0x8003b8dc
[ 5191.111561] [<8003b804>] 0x8003b804
[ 5191.115155] [<80009278>] 0x80009278
[ 5191.118718] 
[ 5191.120239] ---[ end trace acc1e4c71dd06c9e ]---
[ 5213.573322] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.582846] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.592311] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.601821] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.611326] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.620822] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.630280] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.639726] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.649182] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.658637] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.668095] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.677569] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.687034] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.696492] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.705957] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.715419] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.724875] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.734330] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.743784] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.753239] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[ 5213.762694] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
```

ptpt52, 2018-03-02:
hi, is it something related to 2.4G wifi? try to disable wifi 2.4G and test it again. give me some feedback about this.

ptpt52, 2018-03-02:
@droserasprout 
I am sure that this bug is fixed
(WAN interface drops packets heavily)
 https://bugs.lede-project.org/index.php?do=details&task_id=1333

@Mighcavots4 
and it is sure this
`ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2`
is something related to 2.4G wifi

droserasprout, 2018-03-02:
@ptpt52 I've tested the latest build now, seems like the WAN issues are gone. No kernel panics, no huge packet dropping.There's also a huge 5G wi-fi speed gain compared to last working build. Not sure if 2,4G issues are resoved - need more testing - but this ticket is definitely closed.

closed 2018-03-02 by droserasprout