x-wrt/x-wrt #3: R6220 - Unstable WIFI network.

Tempacc129, 2018-03-07:
Hey. R6220 User here.

I tried ur firmware for more tweaks, but ur firmware makes my router WIFI unstable. thus, i want to revert back to original firmware but i dont know how to do that. can u help me please? thanks.

Tempacc129, 2018-03-08:
update : 0306 firmware doesnt works at R6220 - no WAN connection, Wireless cant be turn on'd.

Tempacc129, 2018-03-09:
@ptpt52 i am extreamly stressful coz of this issue. even LAN drops connection frequently so cant do any shits properly. please let me know how to write factory firmware if you can. thanks.

ptpt52, 2018-03-09:
@Tempacc129 
Hi, Is the MAC address and label consistent?
try R6220A R6220B

enable ssh login (system->administrator->enable ssh root login)
get ....r6220a-squashfs-sysupgrade.tar
login ssh
sysupgrade -F -n ....r6220a-squashfs-sysupgrade.tar
check lan MAC address and device label again.

if the MAC address and label still not consistent.
try r6220b again.

Tempacc129, 2018-03-09:
@ptpt52 Thanks for responding! In my case, All R6220 R6220A R6220B Firmwares gave me correct MACs.

I tested them as using LuCI upgrade page - should i try them again with SSH?

Tempacc129, 2018-03-09:
if possible can u also let me know how can i get "tar" extension type of factory firmwares? i failed to get them, instead i got "img" extension file and they were not able to use at LuCI for upgrading.

ptpt52, 2018-03-09:
http://lists.infradead.org/pipermail/lede-commits/2017-March/003257.html
nmrpflash can be used to recover to the netgear firmware if a broken image was flashed.

Tempacc129, 2018-03-09:
@ptpt52 i know that method, but im trying to know how to revert my changes without breaking my router lol..

Tempacc129, 2018-03-09:
and im confused - what actions does break firmware?

ptpt52, 2018-03-09:
use nmrpflash to revert back to facotry firmware?
or
try reset factory setting via holding reset button for 7 seconds while system is running

Tempacc129, 2018-03-09:
nmrpflash only works when router gets bricked(broken) iirc, thus i have to break my router.

that would break my system?

ptpt52, 2018-03-09:
just use nmrpflash anyway to revert back to factory system.
or
pressing reset button for 7sec for factory reset only.

Tempacc129, 2018-03-09:
> nmrpflash only works when router gets bricked(broken) iirc, thus i have to break my router.
> what actions does break firmware?
Why are u suggesting factory reset, then?

ptpt52, 2018-03-31:
@Tempacc129 

> @ptpt52 Thanks for responding! In my case, All R6220 R6220A R6220B Firmwares gave me correct MACs.
I tested them as using LuCI upgrade page - should i try them again with SSH?

it maybe wrong, it can't be 'All R6220 R6220A R6220B Firmwares gave you correct MACs.'
only one firmware is correct.

if the macs is not the same as device LABEL one. you may use wrong firmware.

try different firmware （R6220/R6220A/R6220A）via sysupgrade -n -F /tmp/xxxx-sysupgrade.tar