x-wrt/x-wrt #4: 编译过程遇到 Error

cyrilkong, 2018-03-27:
make[3] -C feeds/packages/net/usbip compile

请问这是缺少了什么 package 吗？

ptpt52, 2018-03-27:
什么问题。。make V=s 看详情

cyrilkong, 2018-03-28:
應該是卡住了在 usbip 的途中

cyrilkong, 2018-03-28:
刷进去 192.168.15.1 但不知道 root 密码是什么

cyrilkong, 2018-03-28:
试过admin试过root试过88888888也是不行，怎么复位firstboot状态？

cyrilkong, 2018-03-28:
我尝试了用failsafe的方法长按reset按钮并设置192.168.1.2，结果可以看到ping通了192.168.1.1，但telnet没有反应

ptpt52, 2018-03-29:
如果是这套自己编译的固件，默认IP 192.168.15.1 启动后，ssh 登录不需要密码，设置密码后，界面可以登录

恢复出厂，捅菊花

如果是我发布的固件：
https://router-sh.ptpt52.com/rom/
配置如下：
固件无线默认名称：NATCAP_XXXX，密码：88888888
固件管理界面：http://192.168.15.1/
管理界面账户/密码：root/admin

cyrilkong, 2018-03-29:
自己编译的，空白我也是试过了，现在情况是：
1. Router LED正常
2. 能进去 192.168.15.1
3. 账号 root 密码全都不对，SSH 同样也是
4. 尝试 Failsafe 菊花长按，电脑设定 192.168.1.2
5. 能够 ping 活 192.168.1.1，但是 telnet 出现 timed out

请问这个情况我该如何恢复原始 ROM 或是重新刷另一个 bin？谢谢dalao

cyrilkong, 2018-03-29:
[image]
目前 Luci 登入不成功

ptpt52, 2018-03-29:
或者你上次升级的时候，没有清空配置，那么密码可能是老的配置。
你尝试 恢复刷回去官方原厂固件，重新开始

cyrilkong, 2018-03-30:
感谢解答问题，目前我这个情况要怎么恢复刷回去官方原厂固件，好像 ssh 和 telnet 都进去不了

ptpt52, 2018-03-30:
什么硬件型号？去谷歌搜索怎样恢复原厂。。。

cyrilkong, 2018-03-30:
谢谢帮忙
Asus ACRH17 (Qualcom ipq4019)
长按 reset 的办法我试过，还是回到 luci 管理页，密码还是不对

ptpt52, 2018-03-30:
我不知道你的固件如何生成，用的什么代码
相关提示：
编译环境里面，密码文件 package/base-files/files/etc/shadow 默认的root密码是空的
编译环境生成的密码 build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/etc/shadow

你的实在没办法，只能 拆开，接串口处理。。。何必自己折腾，用我发布的固件肯定没问题。

shadow 文件：

> root::0:0:99999:7:::
daemon:*:0:0:99999:7:::
ftp:*:0:0:99999:7:::
network:*:0:0:99999:7:::
lede:*:0:0:99999:7:::
nobody:*:0:0:99999:7:::
dnsmasq: x:0:0:99999:7:::
mosquitto: x:0:0:99999:7:::

cyrilkong, 2018-03-30:
我错了，我就是想要加 v2ray 进去，自己编译个简洁的玩的，本来打算 factory.trx 升级之后就可以刷 opboot 了，却没有想到 ssh 都进不去了，还没不死保险就这样
我看了 shadow 大部分密码都是空白对吧，会不会是账号没有root，改成其他的用户，比如lede？

closed 2018-08-05 by ptpt52