x-wrt/x-wrt #5: Any performance figures on ACRH17+openwrt?

xlighting2017, 2018-04-24:
how to measure can be found here:
https://wiki.openwrt.org/doc/howto/benchmark.openssl

I'd like to know if IPQ4018 can outperform BCM4706, or just as good is it...

ptpt52, 2018-04-24:
| | ARMv7 Processor rev 5 (v7l) | 67.03 | ARMv7 Processor rev 5 (v7l) | 67.03 | ARMv7 Processor rev 5 (v7l) | 67.03 | ARMv7 Processor rev 5 (v7l) | 67.03 | Generic DT based system | 1.0.2o | 60384020 | 38773200 | 23244110 | 19534200 | 8952690 | 3219800 | 18619330 | 16037760 | 13873150 | 20.9 | 823.6 64.2 | 70.3 |

ptpt52, 2018-04-24:
> | | ARMv7 Processor rev 5 (v7l) | 67.03 | ARMv7 Processor rev 5 (v7l) | 67.03 | ARMv7 Processor rev 5 (v7l) | 67.03 | ARMv7 Processor rev 5 (v7l) | 67.03 | Generic DT based system | 1.0.2o | [...]

ptpt52, 2018-04-24:
However, Qualcomm’s chips have better software support, and Broadcom is more paralyzed.

ptpt52, 2018-04-24:
OpenSSL 1.0.2o  27 Mar 2018
built on: reproducible build, date unspecified
options:bn(64,32) rc4(ptr,char) des(idx,cisc,16,long) aes(partial) blowfish(ptr) 
compiler: arm-openwrt-linux-muslgnueabi-gcc -I. -I.. -I../include  -fPIC -DOPENSSL_PIC -DOPENSSL_THREADS -D_REENTRANT -DDSO_DLFCN -DHAVE_DLFCN_H -I/mnt/Data/Sources/openwrt/lede-source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/mnt/Data/Sources/openwrt/lede-source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/include -I/mnt/Data/Sources/openwrt/lede-source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/include -I/mnt/Data/Sources/openwrt/lede-source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.3.0_musl_eabi/include/fortify -I/mnt/Data/Sources/openwrt/lede-source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.3.0_musl_eabi/include -DOPENSSL_NO_ERR -DTERMIOS -pipe -mcpu=cortex-a7 -mfpu=neon-vfpv4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/mnt/Data/Sources/openwrt/lede-source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/openssl-1.0.2o:openssl-1.0.2o -O3 -fpic -I/mnt/Data/Sources/openwrt/lede-source/package/libs/openssl/include -ffunction-sections -fdata-sections -fomit-frame-pointer -Wall -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DAES_ASM -DBSAES_ASM -DGHASH_ASM
The 'numbers' are in 1000s of bytes per second processed.
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes
md5               3754.51k    12706.38k    34102.23k    60384.02k    76207.45k
sha1              3834.92k    11698.88k    26420.14k    38773.20k    44570.23k
des cbc           8514.35k     8837.23k     9028.61k     8952.69k     9024.85k
des ede3          3149.53k     3215.10k     3226.45k     3219.80k     3246.76k
aes-128 cbc      16521.48k    17994.28k    18308.76k    18619.33k    18576.73k
aes-192 cbc      14044.60k    15419.88k    15865.99k    16037.76k    15997.69k
aes-256 cbc      12664.59k    13426.80k    13700.76k    13873.15k    13822.98k
sha256            4515.24k    10341.37k    18457.34k    23244.11k    24216.56k
sha512            2372.89k     9494.69k    14128.81k    19534.20k    22041.69k
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.047885s 0.001214s     20.9    823.6
                  sign    verify    sign/s verify/s
dsa 2048 bits 0.015569s 0.014223s     64.2     70.3

xlighting2017, 2018-04-25:
Thanks a lot! I get it :)

closed 2018-04-25 by xlighting2017