x-wrt/x-wrt #8: Wifi Problems with Asus RT-AC58U QCA4019

F1ll0y, 2018-05-15:
I have problem with my routers. This is the error log:

`[   20.581874] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/pre-cal-ahb-a000000.wifi.bin failed with error -2
[   20.581920] ath10k_ahb a000000.wifi: Falling back to user helper
[   20.677763] ubi0 error: vol_cdev_read: damaged volume, update marker is set
[   20.680724] firmware ath10k!pre-cal-ahb-a000000.wifi.bin: firmware_loading_store: map pages failed
[   20.683939] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/cal-ahb-a000000.wifi.bin failed with error -2
[   20.692693] ath10k_ahb a000000.wifi: Falling back to user helper
[   20.735472] firmware ath10k!cal-ahb-a000000.wifi.bin: firmware_loading_store: map pages failed
[   20.735784] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/firmware-6.bin failed with error -2
[   20.743073] ath10k_ahb a000000.wifi: Falling back to user helper
[   20.807441] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   20.809184] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   20.815436] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   20.829291] ath10k_ahb a000000.wifi: firmware ver 10.4-3.4-00104 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 6c332c67
[   20.851504] ath10k_ahb a000000.wifi: failed to fetch board data for bus=ahb,vendor=0000,device=0000,subsystem-vendor=0000,subsystem-device=0000,variant=RT-AC58U from ath10k/QCA4019/hw1.0/board-2.bin
[   20.851761] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/board.bin failed with error -2
[   20.868121] ath10k_ahb a000000.wifi: Falling back to user helper
[   20.911753] firmware ath10k!QCA4019!hw1.0!board.bin: firmware_loading_store: map pages failed
[   20.911974] ath10k_ahb a000000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   20.919290] ath10k_ahb a000000.wifi: failed to fetch board file: -11
[   20.928680] ath10k_ahb a000000.wifi: could not probe fw (-11)
[   21.120691] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/pre-cal-ahb-a800000.wifi.bin failed with error -2
[   21.120741] ath10k_ahb a800000.wifi: Falling back to user helper
[   21.212857] ubi0 error: vol_cdev_read: damaged volume, update marker is set
[   21.215669] firmware ath10k!pre-cal-ahb-a800000.wifi.bin: firmware_loading_store: map pages failed
[   21.218950] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/cal-ahb-a800000.wifi.bin failed with error -2
[   21.227764] ath10k_ahb a800000.wifi: Falling back to user helper
[   21.267872] firmware ath10k!cal-ahb-a800000.wifi.bin: firmware_loading_store: map pages failed
[   21.268260] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/firmware-6.bin failed with error -2
[   21.275488] ath10k_ahb a800000.wifi: Falling back to user helper
[   21.335077] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   21.337063] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   21.343028] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   21.357241] ath10k_ahb a800000.wifi: firmware ver 10.4-3.4-00104 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 6c332c67
[   21.378844] ath10k_ahb a800000.wifi: failed to fetch board data for bus=ahb,vendor=0000,device=0000,subsystem-vendor=0000,subsystem-device=0000,variant=RT-AC58U from ath10k/QCA4019/hw1.0/board-2.bin
[   21.379100] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/board.bin failed with error -2
[   21.395452] ath10k_ahb a800000.wifi: Falling back to user helper
[   21.435616] firmware ath10k!QCA4019!hw1.0!board.bin: firmware_loading_store: map pages failed
[   21.435875] ath10k_ahb a800000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   21.443231] ath10k_ahb a800000.wifi: failed to fetch board file: -11
[   21.452647] ath10k_ahb a800000.wifi: could not probe fw (-11)
`

Can somebody help me?

ptpt52, 2018-05-21:
try my rom?
https://router-sh.ptpt52.com/rom/

zimiao815, 2019-03-01:
me too!!  how do?

ptpt52, 2019-03-01:
> me too!! how do?

what is your rom version?
ac58u work perfect on this repo

ptpt52, 2019-03-13:
tested OK with latest code

closed 2019-03-13 by ptpt52