x-wrt/x-wrt #13: acrh17的分区大小好像有些不太对劲

daiaji, 2018-11-09:
```
df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 6.8M      6.8M         0 100% /rom
tmpfs                   122.5M      1.3M    121.3M   1% /tmp
/dev/ubi0_7              60.7M     92.0K     57.5M   0% /overlay
overlayfs:/overlay       60.7M     92.0K     57.5M   0% /
tmpfs         
```
openwrt-ipq40xx-asus_rt-acrh17-initramfs-factory.trx刚刷上的时候 我看见还有90MB的可用空间
openwrt-ipq40xx-asus_rt-acrh17-squashfs-sysupgrade.bin刷上后就这样了……

ptpt52, 2018-11-09:
这是正确的。

ptpt52, 2018-11-09:
initramfs 刷进去是临时启动在内存里，是无法保存配置的，重启就没有了。
刷sysupgrade才用flash空间做文件系统保存数据
分区：
Factory(124KB),linux(19.9MB),linux2(19.9MB),jffs2(2.4MB),rootfs+rootfs_data(~75MB)
最后真正剩下可以写的空间就是这么多了

daiaji, 2018-11-09:
@ptpt52 linux linux2 是什么鬼东西？
然后你把你下载站的ssl证书整了吧……
自签证书从SSL的安全角度来讲也挺可怕的

ptpt52, 2018-11-09:
两个兼容原厂的内核分区，可以下载官方固件，sysupgrade -n -F RT-ACRH17.trx 刷回原厂。

ssl证书自签名一定程度上安全的，如果有人要中间人攻击冒充我的网站，随便吧。

daiaji, 2018-11-09:
@ptpt52 能不能把openwrt只读的那部分全塞到linux linux2？
这样rootfs就不占data了吧
我看整个固件也就那么点大
实际上不编译很多东西的19M或者 19M*2也够用了

ptpt52, 2018-11-09:
记得了，兼容分区是在系统坏了后，开机嗯reset进入恢复模式，tftp还原原厂系统。

ptpt52, 2018-11-09:
不会改变分区，就这样比较合理。你不需要那么多空间，50M足够用了。

daiaji, 2018-11-09:
@ptpt52 只是rootfs还塞本来就不多的70MB分区里 不太好吧
用raid之类的东西貌似是可以把两个分区合成一个块设备吧（我用btrfs的时候貌似见过不做raid 直接把两个分区合成一个逻辑分区的 小米路由的openwrt上貌似这么搞过 那东西是两块flash）
只读分区塞linux分区就基本完事了
恢复原厂的话是直接写块吧
那openwrt只要不重新分区
怎么都能还原原厂啊

ptpt52, 2018-11-09:
linux 分区已经是kenel了。linux2大小是固定的，不适合做rootfs。rootfs大小是打包的时候选择打包的软件越多空间需要越大，就有可能超过linux2的大小，所以做rootfs也不合理。

closed 2019-03-13 by ptpt52

daiaji, 2020-02-06:
@ptpt52 我这边进官方bl的救援模式后，没法用tftp推固件让他恢复…
但是可以ping通
咋整？