x-wrt/x-wrt #22: M4.14.18

xftv5608, 2019-04-07:
make[7]: *** [Makefile:1910: freadahead.o] Error 1
make[7]: Leaving directory '/home/openwrt/x-wrt/build_dir/host/m4-1.4.18/lib'
make[6]: *** [Makefile:1674: all] Error 2
make[6]: Leaving directory '/home/openwrt/x-wrt/build_dir/host/m4-1.4.18/lib'
make[5]: *** [Makefile:1572: all-recursive] Error 1
make[5]: Leaving directory '/home/openwrt/x-wrt/build_dir/host/m4-1.4.18'
make[4]: *** [Makefile:1528: all] Error 2
make[4]: Leaving directory '/home/openwrt/x-wrt/build_dir/host/m4-1.4.18'
make[3]: *** [Makefile:30: /home/openwrt/x-wrt/build_dir/host/m4-1.4.18/.built] Error 2
make[3]: Leaving directory '/home/openwrt/x-wrt/tools/m4'
time: tools/m4/compile#19.51#8.50#29.00
make[2]: *** [tools/Makefile:161: tools/m4/compile] Error 2
make[2]: Leaving directory '/home/openwrt/x-wrt'
make[1]: *** [tools/Makefile:157: /home/openwrt/x-wrt/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynnyyy] Error 2
make[1]: Leaving directory '/home/openwrt/x-wrt'
make: *** [/home/openwrt/x-wrt/include/toplevel.mk:218：world] 错误 2

xftv5608, 2019-04-07:
freadahead.c:92:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."

ptpt52, 2019-04-12:
what?

closed 2020-03-08 by ptpt52