x-wrt/x-wrt #24: Add Xiaomi Mi R3 to the official OpenWRT

nicovell3, 2019-04-10:
Would it be possible if you submitted a pull request at the official project so the Xiaomi MiWiFi Router 3 is officially supported?

If you don't want to make such job, do you mind if I submit a pull request to add that support?

Thanks in advance!

ptpt52, 2019-04-10:
I did it before https://github.com/openwrt/openwrt/pull/597
but the nand driver for mir3 is dirty, cannot be accepted

you can do it, if u can reuse the mtk-nand driver or something

Goshin, 2020-05-07:
@ptpt52 你好，我想给我的 Mi R3 编译一个 openwrt，仅作个人自用。

我大概看了一下你的仓库源码，想请问一下是不是只要 cherry-pick https://github.com/x-wrt/x-wrt/commit/f8ba2c7b21102d1e7f3654e5bf42dd273dc1baf1 和 https://github.com/x-wrt/x-wrt/commit/bd359c7fe1a57d98339e8f041cbbb32585338968 这两个 commit 再编译就可以了？

因为这两个 commit 的代码跟原先你在原来 PR 里的代码还不太一样，所以才想先联系你发问确认一下。

很感谢你的贡献和分享的代码！👍

Neustradamus, 2021-08-28:
@ptpt52: Can you look for the support of the Xiaomi Mi Router 3?

There were 2 PRs:
- https://github.com/openwrt/openwrt/pull/597
- https://github.com/openwrt/openwrt/pull/764

ptpt52, 2021-08-29:
@Neustradamus 
since the nand driver for mt7620 is dirty and cannot be accepted, openwrt would probably do not support mir3

Neustradamus, 2021-08-29:
@ptpt52: I think it is possible to change it...

Can you look?

cc: @jow-, @nbd168, @adschm, @aparcar, @inflabz, @iscilyas, @sequentiality, @ioiotor, @chunkeey, @ynezz

Several people would like to have the support of the Xiaomi Mi Router 3 like other Xiaomi devices...