x-wrt/x-wrt #25: 多台X86设备实测NFT-QoS静态限速功能无效

lewyue, 2019-04-13:
**Describe the bug**
多台X86设备实测NFT-QoS静态限速功能无效。
测试了两台设备，分别刷的XWRT最新版本，静态限速均无效。
同款设备测试刷了其它的两款X86系统，静态限速功能有效，应该是有BUG

lewyue, 2019-04-13:
补充下，经过反复测试
从电脑到路由，局域网间倒是被限速了，比如上传固件到路由器是限速的
但电脑访问公网的速度，没有限制住

ptpt52, 2019-04-13:
@lewyue 
请在 `服务` `基础系统` 页面 `启用转发加速(natflow)` 关闭 后再测试。

转发加速(natflow) 功能和流量控制功能是冲突的

lewyue, 2019-04-13:
收到，感谢，应该是这个问题，建议在页面上加个说明

LGA1150, 2019-04-19:
开启 flow offload 后，所有已建立的连接会绕过防火墙规则，包括 nftables 的限速策略
此时，要用基于 tc 的限速策略，OpenWrt 源里的其他两个 QoS 均能使用

closed 2019-06-02 by ptpt52

yubanmeiqin9048, 2020-05-26:
可以启用BBR吗

ptpt52, 2020-05-26:
> 可以启用BBR吗

能呀。不过客户端开启这个没什么用，效果不一定好。
`echo net.ipv4.tcp_congestion_control=bbr  >>/etc/sysctl.d/50-local.conf`

yubanmeiqin9048, 2020-05-26:
> > 可以启用BBR吗
> 
> 能呀。不过客户端开启这个没什么用，效果不一定好。
> `echo net.ipv4.tcp_congestion_control=bbr >>/etc/sysctl.d/50-local.conf`

现在遇到的问题就是，限速需要通过重启路由器才能生效

ptpt52, 2020-05-26:
@yubanmeiqin9048  
流量控制并不存在你说的问题，实测生效正常。
注意流量控制功能 一定要关闭natflow功能。

yubanmeiqin9048, 2020-06-13:
> @yubanmeiqin9048
> 流量控制并不存在你说的问题，实测生效正常。
> 注意流量控制功能 一定要关闭natflow功能。

FULLCONE NAT加速需要关闭吗（静态限速）