x-wrt/x-wrt #28: breed-factory.bin如何制作的

feika, 2019-05-13:
想请教下小米的那几个breed-factory.bin是如何制作的，我想自己修改后编译，小米官方的openwrt都没有factory包

ptpt52, 2019-05-13:
修改Makefile对应的定义
在文件 `x-wrt/target/linux/ramips/image/mt7621.mk`

```
define Device/xiaomi_mir3p
  DTS := MIR3P
  BLOCKSIZE := 128k
  PAGESIZE := 2048
  KERNEL_SIZE:= 4096k
  UBINIZE_OPTS := -E 5
  IMAGE_SIZE := $(ralink_default_fw_size_32M)
  DEVICE_TITLE := Xiaomi Mi Router 3 Pro (no wifi)
  IMAGES += kernel1.bin rootfs0.bin breed-factory.bin
  IMAGE/kernel1.bin := append-kernel
  IMAGE/rootfs0.bin := append-ubi | check-size $$$$(IMAGE_SIZE)
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  IMAGE/breed-factory.bin := append-kernel | pad-to $$(KERNEL_SIZE) | \
                             append-kernel | pad-to $$(KERNEL_SIZE) | \
                             append-ubi | check-size $$$$(IMAGE_SIZE)
  DEVICE_PACKAGES := \
    kmod-usb3 kmod-usb-ledtrig-usbport uboot-envtools
endef
TARGET_DEVICES += xiaomi_mir3p

define Device/xiaomi_mir3g
  DTS := MIR3G
  BLOCKSIZE := 128k
  PAGESIZE := 2048
  KERNEL_SIZE := 4096k
  IMAGE_SIZE := 32768k
  UBINIZE_OPTS := -E 5
  IMAGES += kernel1.bin rootfs0.bin breed-factory.bin
  IMAGE/kernel1.bin := append-kernel
  IMAGE/rootfs0.bin := append-ubi | check-size $$$$(IMAGE_SIZE)
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  IMAGE/breed-factory.bin := append-kernel | pad-to $$(KERNEL_SIZE) | \
                             append-kernel | pad-to $$(KERNEL_SIZE) | \
                             append-ubi | check-size $$$$(IMAGE_SIZE)
  DEVICE_TITLE := Xiaomi Mi Router 3G
  SUPPORTED_DEVICES += R3G
  SUPPORTED_DEVICES += mir3g
  DEVICE_PACKAGES := \
    kmod-mt7603 kmod-mt76x2 kmod-usb3 kmod-usb-ledtrig-usbport \
    uboot-envtools
endef
TARGET_DEVICES += xiaomi_mir3g
```

注意它们分区的分布情况，都是 (4m)kernel1+(4m)kernel2+rootfs

feika, 2019-05-14:
谢谢

closed 2019-05-14 by feika