x-wrt/x-wrt #35: 希望能加入HiLink支持

tidusdacapo, 2019-06-22:
雖然changelog上有提及"全面支持4G USB插卡"
但當用上HUAWAI的E3372的時侯
出現了error network device is not present
估計是缺了"kmod-usb-net-cdc-ether"

source: https://protyposis.net/blog/using-the-huawei-e3372-hi-link-lte-dongle-with-openwrt/

ptpt52, 2019-06-22:
并不缺少 kmod-usb-net-cdc-ether 原因还需分析

tidusdacapo, 2019-06-22:
感謝回覆
即使再安裝了usbmodeswitch情況依然一樣
根據搜集得的資料"kmod-mii"也有可能需要

現付上有可能用得上的資料
https://blog.samnya.cn/tag/openwrt/
https://wdicc.com/openwrt/

p,s. 
http://ofmodemsandmen.com/about.html
源碼: https://drive.google.com/file/d/185UpoL8Kjuf6v8ogmfIiOKEyzuTJurfg/view

ptpt52, 2019-06-22:
什么型号的设备？并不是所有设备型号都包含usb 4G相关驱动。

ptpt52, 2019-06-22:
flash <=8M + usb口，支持普通4G手机usb共享（rndis驱动）
flash > 8M + usb口，支持相对更加完整的 4G usb功能。

tidusdacapo, 2019-06-22:
我是用華為E3372的, 純官方系統中能正確使用
而路由就是TP-Link TL-WDR4300 v1
的確是8M flash的

上面的源碼是我過去編譯openwrt使用過, 能很方便的支持大部分4G usb modem
希望能幫到你

ptpt52, 2019-06-23:
8M flash确实不支持这个，是因为空间不够。
不过这个wdr4300空间剩余足够，我考虑加入支持一下。

tidusdacapo, 2019-06-23:
感謝

因香港用戶無法買恩山邀请码
能在這反映到太好了

ptpt52, 2019-06-24:
好的，x-wrt-5.0-b201906240754 包括之后的版本已经加入了支持

closed 2020-03-08 by ptpt52