x-wrt/x-wrt #38: 能否适配一个小米路由器PRO(R3P)得SPI 闪存版

laomao9000, 2019-07-24:
小米路由器pro使用nand闪存，很多朋友刷机报错，变砖，原因就是NAND坏块问题一直解决不好。为了救砖，好在电路板上又SPI闪存的焊盘，可以很方便地改造为SPI闪存启动，只是目前没有定制固件，希望能给予支持。
谢谢！

ptpt52, 2019-07-25:
可以支持，初步分区规划16M flash
```
192k(u-boot),64k(u-boot-env),64k(factory_rsvd),15808k(firmware),256k(Factory)
```
最后一个分区是 Factory 和原来nand上的 Factory分区一样，你需要把原来的Factory备份，写入 Factory
```
mtd write /tmp/Factory.backup.img Factory && reboot
```

ptpt52, 2019-07-25:
u-boot 可以刷 breed-mt7621-pbr-m1.bin 的

laomao9000, 2019-07-25:
感谢！
最好将Factory放到常用位置，方便固件切换
`[    2.224000] PandoraBox MT7621/28 SPI flash driver, SPI_ClK: 31MHz
[    2.228000] device id : ef 40 19 0 0 (40190000)
[    2.232000] SPI flash chip: W25Q256FV (ef 40190000) (32768 Kbytes)
[    2.240000] Creating 5 MTD partitions on "raspi":
[    2.244000] 0x000000000000-0x000000030000 : "u-boot"
[    2.252000] 0x000000030000-0x000000040000 : "u-boot-env"
[    2.260000] 0x000000040000-0x000000050000 : "Factory"
[    2.264000] 0x000000000000-0x000002000000 : "fullflash"
[    2.272000] 0x000000050000-0x000002000000 : "firmware"
`

ptpt52, 2019-07-25:
常用未知的Factory 只有 64K
米3G 原厂的Factory 是256K，大小不同。所以这里放到最后，才有足够空间

laomao9000, 2019-07-25:
前64K就够了，0x8400后面的全是0xFF，没有数据的。只在0xE000保存了MAC地址。

laomao@lvm:~$ hexdump Factory.bin
0000000 7615 0070 1178 05dc 4ec2 0000 0000 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0200
0000020 0000 0000 0000 0000 0000 0000 0000 0000
0000030 0000 0000 0044 6000 0001 0000 0000 0000
0000040 0000 0044 0004 0000 0000 0000 0000 01e0
0000050 2020 b300 b840 c440 c124 c2c1 c3c0 8124
0000060 8182 c4c0 8324 8384 c440 8124 8281 c540
0000070 0014 c000 14c4 0000 c3c0 0014 4000 14c3
0000080 0000 c240 0014 4000 14c2 0000 c240 0014
0000090 c000 14c2 0000 c340 0014 4000 14c3 0000
00000a0 c340 0014 c000 14c2 0000 c240 0014 c000
00000b0 14c1 0000 c140 0014 c000 14c0 0000 c6c6
00000c0 c4c4 00c4 c200 c200 81c1 8181 c285 00c2
00000d0 8282 8182 c500 c5c5 0000 00c3 c3c3 8282
00000e0 8382 c384 c3c3 8282 0082 8200 0082 0000
00000f0 0085 0000 00c1 8585 0000 0000 0000 8600
0000100 0000 0000 0000 0000 0000 0000 0000 0000
0000110 0000 0000 0000 0000 00c6 0000 0000 0000
0000120 0000 0000 0000 0000 0000 0000 0000 0000
*
0000140 c140 0014 c000 14c1 0000 c140 0014 4000
0000150 14c1 0000 c140 0014 c000 14c0 0000 c0c0
0000160 0014 c000 14c0 0000 c5c0 0014 4000 14c5
0000170 0000 c440 0014 c000 14c3 0000 c340 0014
0000180 c000 14c2 0000 c2c0 0014 c000 14c2 0000
0000190 0000 0000 0000 0000 0000 0000 0000 0000
*
0000220 0300 001e bd05 0500 005d 1707 ba14 0051
0000230 0000 0000 0000 0000 0000 0000 0000 0000
*
00003a0 0000 0000 0000 0000 0000 0000 0307 03f1
00003b0 03f1 00f1 0000 0000 0000 0000 0000 0000
00003c0 0000 0000 0000 0000 0000 0000 0000 0000
*
0000400 ffff ffff ffff ffff ffff ffff ffff ffff
*
0008000 7615 0070 1178 05dc 4fc2 0000 0000 0000
0008010 0000 0000 0000 0000 0000 0000 0000 0200
0008020 0000 0000 0000 0000 0000 0000 0000 0000
0008030 0000 0000 0044 6000 0001 0000 0000 0000
0008040 0000 0044 0004 0000 0000 0000 0000 01e0
0008050 2020 b300 b840 c440 0014 0000 c3c0 0014
0008060 0000 c4c0 0014 0000 c440 0014 0000 c540
0008070 0020 c000 22c4 8080 c3c0 0020 4000 20c3
0008080 0000 c240 c422 40c4 20c2 0000 c240 c522
0008090 c0c5 22c2 c3c3 c340 0014 4000 22c3 c1c1
00080a0 c340 0014 c000 14c2 0000 c240 c222 c0c2
00080b0 14c1 0000 c140 c522 c0c5 22c0 c4c4 c6c6
00080c0 c4c4 00c4 c200 c200 81c1 8181 c285 00c2
00080d0 8282 8182 c500 c5c5 0000 00c3 c3c3 8282
00080e0 8382 c384 c3c3 8282 0082 8200 0082 0000
00080f0 0085 0000 00c1 8785 0000 0000 0000 8600
0008100 0000 0000 0000 0000 0000 0000 0000 0000
0008110 0000 0000 0000 0000 00c5 0000 0000 0000
0008120 0000 0000 0000 0000 0000 0000 0000 0000
*
0008140 c140 0014 c000 22c1 8080 c140 0014 4000
0008150 14c1 0000 c140 c122 c0c1 14c0 0000 c0c0
0008160 c222 c0c2 22c0 c2c2 c5c0 0014 4000 22c5
0008170 8383 c440 0014 c000 14c3 0000 c340 c222
0008180 c0c2 14c2 0000 c2c0 c422 c0c4 22c2 c3c3
0008190 0000 0000 0000 0000 0000 0000 0000 0000
*
0008250 0000 0000 fa03 0300 005d 9b02 0500 0001
0008260 0000 0000 ea00 894e 0000 0000 0000 0000
0008270 0000 0000 0000 0000 0000 0000 0000 0000
*
00082a0 0000 0000 0000 0000 0000 1105 0500 005e
00082b0 9705 1000 00fd 0000 0000 fc00 8855 0000
00082c0 0000 0000 0000 0000 0000 0000 0000 0000
*
00082e0 0000 0000 0000 2819 0e00 006b a210 0c00
00082f0 0010 0000 0000 0b00 885a 0000 0000 0000
0008300 0000 0000 0000 0000 0000 0000 0000 0000
*
0008340 1a00 0030 5813 1500 0087 110e 0000 0000
0008350 0000 6612 0091 0000 0000 0000 0000 0000
0008360 0000 0000 0000 0000 0000 0000 0000 0000
*
00083a0 0000 0000 0000 0000 0000 0000 0307 03f1
00083b0 03f1 00f1 0000 0000 0000 0000 0000 0000
00083c0 0000 0000 0000 0000 0000 0000 0000 0000
*
0008400 ffff ffff ffff ffff ffff ffff ffff ffff
*
000e000 1178 05dc 4cc2 1178 05dc 4dc2 ffff ffff
000e010 ffff ffff ffff ffff ffff ffff ffff ffff
*
0040000

ptpt52, 2019-07-25:
也有道理，哈。

laomao9000, 2019-07-25:
> 也有道理，哈。

MT7615用的是开源驱动吗？这样可以开通802.11r了。

ptpt52, 2019-07-25:
刚刚发了个版，支持了，测试反馈一下哈。

laomao9000, 2019-07-26:
已刷机，测试中。
LAN正常，2G和5Gwifi启动慢，要过好久才能搜到信号。
还有，没有酸酸乳，哪里有OPKG源？

ptpt52, 2019-07-26:
界面开启ssh，命令操作 opkg update opkg install xxx
其他的，没有。

ptpt52, 2019-07-26:
dmesg 信息发出来看看
无线，好像需要手动设置 信道，才能工作，目前驱动可能有bug

ZombieFanHH, 2019-07-28:
请教一下刷了Pandorabox 的小米R3P 可以刷这个吗？

ptpt52, 2019-07-29:
@ZombieFanHH 
要么先刷回原厂，然后参考看小米3G的教程刷
要么看情况了，我不知道pandorabox怎样分区不好说

ZombieFanHH, 2019-07-30:
好的，谢谢

daiaji, 2019-08-25:
NAND这么容易坏块吗？
ROM又不是天天刷
差不多是基本只读……

closed 2020-03-08 by ptpt52