x-wrt/x-wrt #40: R7800与intel 9560ac

YuhaoTang, 2019-07-31:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

ptpt52, 2019-08-01:
这个没研究过。有没有测试更高性能的案例？

YuhaoTang, 2019-08-01:
> 这个没研究过。有没有测试更高性能的案例？

原厂固件无线上传下载都能稳定800Mbps以上，而且如果在路由模式下没有任何问题，但我需要使用有线AP模式，此模式下跑满带宽一会儿（10分钟不到吧）就会迷之断网
而x-wrt是少数（甚至我用过的里面的唯一）能和9560ac协商达到1.7Gbps的第三方固件，并且稳定不掉线，唯一的问题就是1.7Gbps像假的一样

ptpt52, 2019-08-01:
cpu 使用情况如何？另外开一个ssh，执行 `mpstat -P ALL 1` 观察cpu情况，再跑iperf 测试

YuhaoTang, 2019-08-02:
首先确实是cpu单核满了，很有可能是这个导致速度上不去，但是更加懵逼的情况出现了，下载会突然跳到750+，突然降到350+，中间没有任何其他操作，网卡协商速度也维持在1.5Gbps左右，但这两个实际速度之间没有其他速度，完全是跳变。录了一段视频，用H265压缩了一下，见附件。
[out.zip](https://github.com/x-wrt/x-wrt/files/3461560/out.zip)

YuhaoTang, 2019-08-03:
备注一下，视频左侧是远程桌面，用千兆网线直连的路由器，右侧是笔记本，用wifi连的路由器

ptpt52, 2019-08-03:
看起来确实是cpu用完了。在服务，系统优化页面，启用转发加速(natflow) 这个应该是默认开启了吧
这个不开启可能还要跑更低点

YuhaoTang, 2019-08-08:
前几天忘记了，刚才用top看了下，sirq高，ksoftirqd高，转发加速开不开好像没啥影响

MeIsReallyBa, 2020-04-11:
感觉像是cpu负载的问题，网卡有时候和无线跑到一个cpu上速度就会降下来。可以调下irq