x-wrt/x-wrt #41: 内网配置咨询

Guxing0321, 2019-08-22:
Hi Team,

我想问下固件里面默认内网的配置使用的是静态ip模式，这个跟我们正常使用的dncp有啥区别不？
如果我想改成正常的dhcp模式，限制dhcp地址池需要怎么设置？

另外，我使用的是小米3路由器，刷了x-wrt固件后发现，有时候会出现无线连接上去无法上网，但是有线是正常的，这个有什么比较好的处理办法不。

BR
Robin

ptpt52, 2019-08-22:
你可能有所误解，内网设置是内网的IP设置。你所说的DHCP服务设置，是另外一个东西，在 网络，接口，lan口的配置，最下面。

小米3 这个不好说，不知道无线是否有问题

Guxing0321, 2019-08-24:
Hi Team,

我找了一个截图，跟你们确认下，是不是在这边,但是我没有找到配置地址池的地方
[image]

另外，我看到配置静态ip的地方可以选择dhcp客户端，这个是用来做什么的
[image]

无线的问题，我更新了8月19 的固件再观察一段时间

BR
Robin

daiaji, 2019-08-25:
@Guxing0321 Lan口妥妥的是固定IP啊……
Wan口有时候才需要DHCP客户端模式接受来自光猫分配的IP

Guxing0321, 2019-09-02:
Hi daiaji,

我大概明白这个意思了，感谢

最近我朋友反馈一个问题，就是连接wifi，第一次没什么问题，
然后，断开再连就是获取ip，至少一分钟，有时候还会连接失败，得连好几次，有时时候根本连不上，得重启整个路由，这个是为何，是我哪里配置不对？？

[image]

ptpt52, 2019-09-02:
更新个版本再看看，不知道无线2.4G是否有问题

closed 2019-10-17 by ptpt52