x-wrt/x-wrt #45: 小米naono upnp无效

ponnron91766, 2019-09-16:
你好，upnp无法自动添加映射，怎么设置都无法添加，是已知问题吗，大概多久修复。

ptpt52, 2019-09-16:
怎么证明不工作？

ponnron91766, 2019-09-16:
Tue Sep 17 05:01:58 2019 daemon.err miniupnpd[4263]: could not open lease file: /var/run/miniupnpd.leases
Tue Sep 17 05:01:58 2019 daemon.notice miniupnpd[4263]: HTTP listening on port 5000
Tue Sep 17 05:01:58 2019 daemon.warn miniupnpd[4263]: no HTTP IPv6 address, disabling IPv6
Tue Sep 17 05:01:58 2019 daemon.notice miniupnpd[4263]: Listening for NAT-PMP/PCP traffic on port 5351
Tue Sep 17 05:01:58 2019 daemon.err miniupnpd[4263]: PCPSendUnsolicitedAnnounce() IPv6 sendto(): Bad file descriptor
Tue Sep 17 05:03:17 2019 daemon.err miniupnpd[4263]: upnp_event_recv: recv(): Connection reset by peer
Tue Sep 17 05:03:17 2019 daemon.err miniupnpd[4263]: upnpevents_processfds: 0x428ef0, remove subscriber uuid:f8a1600a-50a3-4a11-9b0d-0c578bb279cc after an ERROR cb: http://192.168.1.2:2869/upnp/eventing/gfsvzpxsug
Tue Sep 17 05:03:17 2019 daemon.notice miniupnpd[4263]: upnp_event_send: send(192.168.1.2:2869): Connection reset by peer
Tue Sep 17 05:03:17 2019 daemon.err miniupnpd[4263]: upnpevents_processfds: 0x428ef0, remove subscriber uuid:ca62208c-053b-4345-ba3f-7705e5be55db after an ERROR cb: http://192.168.1.2:2869/upnp/eventing/sltimxrwqp
Tue Sep 17 05:03:31 2019 daemon.warn miniupnpd[4263]: upnp_event_process_notify: connect(192.168.1.2:2869): Connection reset by peer
Tue Sep 17 05:03:31 2019 daemon.err miniupnpd[4263]: upnpevents_processfds: 0x428640, remove subscriber uuid:a9b2bdb0-1020-462c-9c6b-94c5f07b8c58 after an ERROR cb: http://192.168.1.2:2869/upnp/eventing/mjhekulmqy
Tue Sep 17 05:03:53 2019 daemon.err miniupnpd[4263]: upnp_event_recv: recv(): Connection reset by peer
Tue Sep 17 05:03:53 2019 daemon.err miniupnpd[4263]: upnpevents_processfds: 0x428ef0, remove subscriber uuid:0cc851d5-651d-4e5a-ad2f-77d59ae6178b after an ERROR cb: http://192.168.1.2:2869/upnp/eventing/sthnintabq
Tue Sep 17 05:03:53 2019 daemon.err miniupnpd[4263]: upnp_event_recv: recv(): Connection reset by peer
Tue Sep 17 05:03:53 2019 daemon.err miniupnpd[4263]: upnpevents_processfds: 0x428ef0, remove subscriber uuid:fbf96ecc-5de0-4178-a118-ccab8e41bdbf after an ERROR cb: http://192.168.1.2:2869/upnp/eventing/jtjfcdxmgp
[image]
[image]

ponnron91766, 2019-09-16:
[image]

ponnron91766, 2019-09-16:
是什么问题吗，大佬

ptpt52, 2019-09-16:
我这边是工作正常的，默认配置，开启就行了。没改过什么

ptpt52, 2019-09-16:
[image]

ponnron91766, 2019-09-16:
emmmm..我也没改过什么，刷好了就直接用，用原版固件upnp可以用，这个固件不知道为什么用不了，难受

ptpt52, 2019-09-16:
iptables-save 命令执行一下看看

ponnron91766, 2019-09-16:
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:18:54 2019
*nat
:PREROUTING ACCEPT [436:35039]
:INPUT ACCEPT [66:4650]
:OUTPUT ACCEPT [184:12643]
:POSTROUTING ACCEPT [57:4510]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i br-lan -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i eth0.2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o br-lan -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o eth0.2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A zone_lan_postrouting -m comment --comment "!fw3: Custom lan postrouting rule chain" -j postrouting_lan_rule
-A zone_lan_prerouting -m comment --comment "!fw3: Custom lan prerouting rule chain" -j prerouting_lan_rule
-A zone_wan_postrouting -m comment --comment "!fw3: Custom wan postrouting rule chain" -j postrouting_wan_rule
-A zone_wan_postrouting -p udp -m comment --comment "!fw3" -j MASQUERADE --random-fully
-A zone_wan_postrouting -j MINIUPNPD-POSTROUTING
-A zone_wan_postrouting -m comment --comment "!fw3" -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "!fw3: Custom wan prerouting rule chain" -j prerouting_wan_rule
-A zone_wan_prerouting -j MINIUPNPD
COMMIT
# Completed on Tue Sep 17 05:18:54 2019
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:18:54 2019
*raw
:PREROUTING ACCEPT [7863:2603424]
:OUTPUT ACCEPT [4537:4009555]
:zone_lan_helper - [0:0]
-A PREROUTING -i br-lan -m comment --comment "!fw3: lan CT helper assignment" -j zone_lan_helper
-A zone_lan_helper -p udp -m comment --comment "!fw3: Amanda backup and archiving proto" -m udp --dport 10080 -j CT --helper amanda
-A zone_lan_helper -p tcp -m comment --comment "!fw3: FTP passive connection tracking" -m tcp --dport 21 -j CT --helper ftp
-A zone_lan_helper -p udp -m comment --comment "!fw3: RAS proto tracking" -m udp --dport 1719 -j CT --helper RAS
-A zone_lan_helper -p tcp -m comment --comment "!fw3: Q.931 proto tracking" -m tcp --dport 1720 -j CT --helper Q.931
-A zone_lan_helper -p tcp -m comment --comment "!fw3: IRC DCC connection tracking" -m tcp --dport 6667 -j CT --helper irc
-A zone_lan_helper -p tcp -m comment --comment "!fw3: PPTP VPN connection tracking" -m tcp --dport 1723 -j CT --helper pptp
-A zone_lan_helper -p tcp -m comment --comment "!fw3: SIP VoIP connection tracking" -m tcp --dport 5060 -j CT --helper sip
-A zone_lan_helper -p udp -m comment --comment "!fw3: SIP VoIP connection tracking" -m udp --dport 5060 -j CT --helper sip
-A zone_lan_helper -p udp -m comment --comment "!fw3: SNMP monitoring connection tracking" -m udp --dport 161 -j CT --helper snmp
-A zone_lan_helper -p udp -m comment --comment "!fw3: TFTP connection tracking" -m udp --dport 69 -j CT --helper tftp
-A zone_lan_helper -p tcp -m comment --comment "!fw3: RTSP connection tracking" -m tcp --dport 554 -j CT --helper rtsp
COMMIT
# Completed on Tue Sep 17 05:18:54 2019
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:18:54 2019
*mangle
:PREROUTING ACCEPT [4060:1118464]
:INPUT ACCEPT [3243:885379]
:FORWARD ACCEPT [762:227382]
:OUTPUT ACCEPT [4544:4012107]
:POSTROUTING ACCEPT [5281:4238441]
-A FORWARD -o br-lan -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone lan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
COMMIT
# Completed on Tue Sep 17 05:18:54 2019
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:18:54 2019
*filter
:INPUT ACCEPT [0:0]
:FORWARD DROP [0:0]
:OUTPUT ACCEPT [0:0]
:MINIUPNPD - [0:0]
[... 55 lines cut ...]
-A zone_lan_input -m comment --comment "!fw3" -j zone_lan_src_ACCEPT
-A zone_lan_output -m comment --comment "!fw3: Custom lan output rule chain" -j output_lan_rule
-A zone_lan_output -m comment --comment "!fw3" -j zone_lan_dest_ACCEPT
-A zone_lan_src_ACCEPT -i br-lan -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth0.2 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth0.2 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_DROP -o eth0.2 -m comment --comment "!fw3" -j DROP
-A zone_wan_dest_REJECT -o eth0.2 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -j MINIUPNPD
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
[... cut ...]
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i eth0.2 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Tue Sep 17 05:18:54 2019

ptpt52, 2019-09-16:
你是不是  opkg update opkg upgrade升级了？

ponnron91766, 2019-09-16:
这个是刚刚opkg了

ptpt52, 2019-09-16:
不要upgrade 吧，你的miniupnpd 不正确

ponnron91766, 2019-09-16:
之前的也不行

ponnron91766, 2019-09-16:
所以就看看upgrade有没有用

ptpt52, 2019-09-16:
你的规则是这样：（错误的顺序，可能是安装了其他版本的miniupnpd导致的）
```
-A zone_wan_postrouting -m comment --comment "!fw3: Custom wan postrouting rule chain" -j postrouting_wan_rule
-A zone_wan_postrouting -p udp -m comment --comment "!fw3" -j MASQUERADE --random-fully
-A zone_wan_postrouting -j MINIUPNPD-POSTROUTING
-A zone_wan_postrouting -m comment --comment "!fw3" -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "!fw3: Custom wan prerouting rule chain" -j prerouting_wan_rule
```
正确的规则：
```
-A zone_wan_postrouting -m comment --comment "!fw3: Custom wan postrouting rule chain" -j postrouting_wan_rule
-A zone_wan_postrouting -j MINIUPNPD-POSTROUTING
-A zone_wan_postrouting -p udp -m comment --comment "!fw3" -j MASQUERADE --random-fully
-A zone_wan_postrouting -m comment --comment "!fw3" -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "!fw3: Custom wan prerouting rule chain" -j prerouting_wan_rule
```

ptpt52, 2019-09-16:
用的哪个版本的固件包？

ponnron91766, 2019-09-16:
x-wrt-5.0-b201909080437-ramips-mt76x8-xiaomi_miwifi-nano-squashfs-sysupgrade

ponnron91766, 2019-09-16:
x-wrt-5.0-b201909150942-ramips-mt76x8-xiaomi_miwifi-nano-squashfs-sysupgrade

ptpt52, 2019-09-16:
看看这个文件
```
 cat /usr/share/miniupnpd/firewall.include
```

ponnron91766, 2019-09-16:
#!/bin/sh
# miniupnpd integration for firewall3

IPTABLES=/usr/sbin/iptables
IP6TABLES=/usr/sbin/ip6tables

$IPTABLES -t filter -N MINIUPNPD 2>/dev/null
$IPTABLES -t nat -N MINIUPNPD 2>/dev/null
$IPTABLES -t nat -N MINIUPNPD-POSTROUTING 2>/dev/null

[ -x $IP6TABLES ] && $IP6TABLES -t filter -N MINIUPNPD 2>/dev/null

. /lib/functions/network.sh

# helper to insert in chain as penultimate
iptables_prepend_rule() {
	local iptables="$1"
	local table="$2"
	local chain="$3"
	local target="$4"

	$iptables -t "$table" -I "$chain" $($iptables -t "$table" --line-numbers -nL "$chain" | \
		sed -ne '$s/[^0-9].*//p') -j "$target"
}

ADDED=0

add_extzone_rules() {
    local ext_zone="$1"

    [ -z "$ext_zone" ] && return

    # IPv4 - due to NAT, need to add both to nat and filter table
    # need to insert as penultimate rule for forward & postrouting since final rule might be a fw3 REJECT
    iptables_prepend_rule "$IPTABLES" filter "zone_${ext_zone}_forward" MINIUPNPD
    $IPTABLES -t nat -A "zone_${ext_zone}_prerouting"  -j MINIUPNPD
    iptables_prepend_rule "$IPTABLES" nat "zone_${ext_zone}_postrouting" MINIUPNPD-POSTROUTING

    # IPv6 if available - filter only
    [ -x $IP6TABLES ] && {
	iptables_prepend_rule "$IP6TABLES" filter "zone_${ext_zone}_forward" MINIUPNPD
    }
    ADDED=$(($ADDED + 1))
}

# By default, user configuration is king.

for ext_iface in $(uci -q get upnpd.config.external_iface); do
    add_extzone_rules $(fw3 -q network "$ext_iface")
done

add_extzone_rules $(uci -q get upnpd.config.external_zone)

[ "$ADDED" -ne 0 ] && exit 0

# If really nothing is available, resort to network_find_wan{,6} and
# assume external interfaces all have same firewall zone.

# (This heuristic may fail horribly, in case of e.g. multihoming, so
# please set external_zone in that case!)

network_find_wan wan_iface
network_find_wan6 wan6_iface

for ext_iface in $wan_iface $wan6_iface; do
    # fw3 -q network fails on sub-interfaces => map to device first
    network_get_device ext_device $ext_iface
    add_extzone_rules $(fw3 -q device "$ext_device")
done

ptpt52, 2019-09-16:
很明显，你这个文件是错误的

ponnron91766, 2019-09-16:
我重新刷回固件看看

ptpt52, 2019-09-16:
反正我发布的固件，或者完全从这个源码编译，是没有问题的，不知道你改了什么
总之miniupnpd 不能错误

ponnron91766, 2019-09-16:
#!/bin/sh
# miniupnpd integration for firewall3

IPTABLES=/usr/sbin/iptables
IP6TABLES=/usr/sbin/ip6tables

$IPTABLES -t filter -N MINIUPNPD 2>/dev/null
$IPTABLES -t nat -N MINIUPNPD 2>/dev/null
$IPTABLES -t nat -N MINIUPNPD-POSTROUTING 2>/dev/null

[ -x $IP6TABLES ] && $IP6TABLES -t filter -N MINIUPNPD 2>/dev/null

. /lib/functions/network.sh

# helper to insert in chain as penultimate
iptables_prepend_rule() {
	local iptables="$1"
	local table="$2"
	local chain="$3"
	local target="$4"

	NR=`$iptables -t "$table" --line-numbers -nL "$chain"; $iptables -t "$table" --line-numbers -nL "$chain" | grep ".*MASQUERADE.*udp.*random-fully.*"`
	NR=`echo "$NR" | sed -ne '$s/[^0-9].*//p'`

	$iptables -t "$table" -I "$chain" $NR -j "$target"
}

ADDED=0

add_extzone_rules() {
    local ext_zone="$1"

    [ -z "$ext_zone" ] && return

    # IPv4 - due to NAT, need to add both to nat and filter table
    # need to insert as penultimate rule for forward & postrouting since final rule might be a fw3 REJECT
    iptables_prepend_rule "$IPTABLES" filter "zone_${ext_zone}_forward" MINIUPNPD
    $IPTABLES -t nat -A "zone_${ext_zone}_prerouting"  -j MINIUPNPD
    iptables_prepend_rule "$IPTABLES" nat "zone_${ext_zone}_postrouting" MINIUPNPD-POSTROUTING

    # IPv6 if available - filter only
    [ -x $IP6TABLES ] && {
	iptables_prepend_rule "$IP6TABLES" filter "zone_${ext_zone}_forward" MINIUPNPD
    }
    ADDED=$(($ADDED + 1))
}

# By default, user configuration is king.

for ext_iface in $(uci -q get upnpd.config.external_iface); do
    add_extzone_rules $(fw3 -q network "$ext_iface")
done

add_extzone_rules $(uci -q get upnpd.config.external_zone)

[ "$ADDED" -ne 0 ] && exit 0

# If really nothing is available, resort to network_find_wan{,6} and
# assume external interfaces all have same firewall zone.

# (This heuristic may fail horribly, in case of e.g. multihoming, so
# please set external_zone in that case!)

network_find_wan wan_iface
network_find_wan6 wan6_iface

for ext_iface in $wan_iface $wan6_iface; do
    # fw3 -q network fails on sub-interfaces => map to device first
    network_get_device ext_device $ext_iface
    add_extzone_rules $(fw3 -q device "$ext_device")

ponnron91766, 2019-09-16:
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:39:50 2019
*nat
:PREROUTING ACCEPT [146:16102]
:INPUT ACCEPT [47:3153]
:OUTPUT ACCEPT [57:4447]
:POSTROUTING ACCEPT [22:2177]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i br-lan -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i eth0.2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o br-lan -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o eth0.2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A zone_lan_postrouting -m comment --comment "!fw3: Custom lan postrouting rule chain" -j postrouting_lan_rule
-A zone_lan_prerouting -m comment --comment "!fw3: Custom lan prerouting rule chain" -j prerouting_lan_rule
-A zone_wan_postrouting -m comment --comment "!fw3: Custom wan postrouting rule chain" -j postrouting_wan_rule
-A zone_wan_postrouting -j MINIUPNPD-POSTROUTING
-A zone_wan_postrouting -p udp -m comment --comment "!fw3" -j MASQUERADE --random-fully
-A zone_wan_postrouting -m comment --comment "!fw3" -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "!fw3: Custom wan prerouting rule chain" -j prerouting_wan_rule
-A zone_wan_prerouting -j MINIUPNPD
COMMIT
# Completed on Tue Sep 17 05:39:50 2019
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:39:50 2019
*raw
:PREROUTING ACCEPT [2250:530348]
:OUTPUT ACCEPT [1437:632302]
:zone_lan_helper - [0:0]
-A PREROUTING -i br-lan -m comment --comment "!fw3: lan CT helper assignment" -j zone_lan_helper
-A zone_lan_helper -p udp -m comment --comment "!fw3: Amanda backup and archiving proto" -m udp --dport 10080 -j CT --helper amanda
-A zone_lan_helper -p tcp -m comment --comment "!fw3: FTP passive connection tracking" -m tcp --dport 21 -j CT --helper ftp
-A zone_lan_helper -p udp -m comment --comment "!fw3: RAS proto tracking" -m udp --dport 1719 -j CT --helper RAS
-A zone_lan_helper -p tcp -m comment --comment "!fw3: Q.931 proto tracking" -m tcp --dport 1720 -j CT --helper Q.931
-A zone_lan_helper -p tcp -m comment --comment "!fw3: IRC DCC connection tracking" -m tcp --dport 6667 -j CT --helper irc
-A zone_lan_helper -p tcp -m comment --comment "!fw3: PPTP VPN connection tracking" -m tcp --dport 1723 -j CT --helper pptp
-A zone_lan_helper -p tcp -m comment --comment "!fw3: SIP VoIP connection tracking" -m tcp --dport 5060 -j CT --helper sip
-A zone_lan_helper -p udp -m comment --comment "!fw3: SIP VoIP connection tracking" -m udp --dport 5060 -j CT --helper sip
-A zone_lan_helper -p udp -m comment --comment "!fw3: SNMP monitoring connection tracking" -m udp --dport 161 -j CT --helper snmp
-A zone_lan_helper -p udp -m comment --comment "!fw3: TFTP connection tracking" -m udp --dport 69 -j CT --helper tftp
-A zone_lan_helper -p tcp -m comment --comment "!fw3: RTSP connection tracking" -m tcp --dport 554 -j CT --helper rtsp
COMMIT
# Completed on Tue Sep 17 05:39:50 2019
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:39:50 2019
*mangle
:PREROUTING ACCEPT [1921:364442]
:INPUT ACCEPT [943:107310]
:FORWARD ACCEPT [978:257132]
:OUTPUT ACCEPT [1442:634470]
:POSTROUTING ACCEPT [2421:891654]
-A FORWARD -o br-lan -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone lan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
COMMIT
# Completed on Tue Sep 17 05:39:51 2019
# Generated by iptables-save v1.8.3 on Tue Sep 17 05:39:51 2019
*filter
:INPUT ACCEPT [0:0]
:FORWARD DROP [0:0]
:OUTPUT ACCEPT [0:0]
:MINIUPNPD - [0:0]
[... 52 lines cut ...]
-A zone_lan_input -m comment --comment "!fw3" -j zone_lan_src_ACCEPT
-A zone_lan_output -m comment --comment "!fw3: Custom lan output rule chain" -j output_lan_rule
-A zone_lan_output -m comment --comment "!fw3" -j zone_lan_dest_ACCEPT
-A zone_lan_src_ACCEPT -i br-lan -m conntrack --ctstate NEW,UNTRACKED -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth0.2 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth0.2 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_DROP -o eth0.2 -m comment --comment "!fw3" -j DROP
-A zone_wan_dest_REJECT -o eth0.2 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -j MINIUPNPD
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
[... cut ...]
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i eth0.2 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Tue Sep 17 05:39:51 2019

ponnron91766, 2019-09-16:
Tue Sep 17 05:41:29 2019 daemon.err miniupnpd[4341]: upnp_event_recv: recv(): Connection reset by peer
Tue Sep 17 05:41:29 2019 daemon.err miniupnpd[4341]: upnpevents_processfds: 0x428ef0, remove subscriber uuid:291c78ce-cf87-4b30-b3bf-3fa7969c57d2 after an ERROR cb: http://192.168.15.238:2869/upnp/eventing/tnussmglox
Tue Sep 17 05:41:29 2019 daemon.notice miniupnpd[4341]: upnp_event_send: send(192.168.15.238:2869): Connection reset by peer
Tue Sep 17 05:41:29 2019 daemon.err miniupnpd[4341]: upnpevents_processfds: 0x428ef0, remove subscriber uuid:d4c488ae-40da-4cd9-842d-e4f81c0de439 after an ERROR cb: http://192.168.15.238:2869/upnp/eventing/raunlwncma

ptpt52, 2019-09-16:
用什么工具测试？

ponnron91766, 2019-09-16:
现在是什么都没改过的

ponnron91766, 2019-09-16:
迅雷，BitComet，都没有

ptpt52, 2019-09-17:
我也用了BitComet，没有问题，不知道你还有什么不一样

ponnron91766, 2019-09-17:
emmmm,这我就不知道了，我还是刷回原版openwrt，到时候再看看。

ptpt52, 2019-09-17:
具体是哪一个版本的openwrt，下载路径发一下

ponnron91766, 2019-09-17:
https://mirror.kumi.systems/openwrt/releases/18.06.4/targets/
就这个

ptpt52, 2019-11-19:
X-WRT 发布新版本: 6.0_b201911201024
1. 解决拨号上网1-2天后断网的问题
2. 解决UPNP突然不生效问题
3. 解决全锥形NAT功能突然失效问题
4. 同步OpenWrt官方最新代码
5. 更新内核版本，部分设备型号的驱动
6. 竞斗云采用ath10k驱动提高5G性能

请移步官网下载 https://x-wrt.com/

ponnron91766, 2019-11-20:
[image]
emmm

daiaji, 2019-12-19:
@ptpt52 竞斗云默认没用ath10k吗？后面才改用的？