x-wrt/x-wrt #48: 多拨配置问题咨询

Guxing0321, 2019-09-24:
Hi Team,

我按照官网的指示配置多拨，但是发现报错

[image]

我看了下接口的配置如下：

[image]

能帮我看看是不是我哪里配置的有问题， 感谢！

BR
Robin

Guxing0321, 2019-09-24:
Hi Team,

如果需要提供相应的配置文件或者日志文件做分析的话，请及时告知我,，谢谢~

BR
Robin

ptpt52, 2019-09-24:
多拨是否支持，看不同地区不同运营商的情况，你那个不行就说明运营商不支持啦

Guxing0321, 2019-09-27:
Hi Team,

关于多拨近期发现一个很神奇的现象，
我之前配置了多拨次数为2，没有起作用，因为不影响上网，就放着没管它

然后，突然某天回去一不小心发现另外一个xwan接口也连上了，而且确实速度快了很多。但是，当我尝试修改多拨次数为4或者改回来之后，发现又都不行了。

我查过一些文档，发现，好像openwrt 的固件有些配置在网页端配的不是很靠谱， 我想咨询下：
我这边可以通过ssh 到机器后台怎么去查看相关的配置是不是正常（配置文件的路径名字等），通过哪个脚本来重启多拨相关服务？

BR
Robin

ptpt52, 2019-09-27:
开启ssh https://blog.x-wrt.com/docs/ssh-open/

意外发现支持多拨，并不代表你的环境支持。目前系统的多拨是传统的方式。
据说有另外一种，需要拨号程序同时并发拨号，这个方式，在这个系统是不支持的，目前也没打算支持。

mlpzaq9998, 2019-10-12:
并发拨号 期待能支持起来 ，目前能用的只有高恪

closed 2020-03-08 by ptpt52