x-wrt/x-wrt #49: K3最近版本5Gwifi驱动一直有问题

xztxy, 2019-09-26:
麻烦大佬看下，K3的固件 5G的wifi出不来，设置后出现“”无线未关联“”
[image]

ptpt52, 2019-09-26:
哪个版本没有问题，大概日期？

xztxy, 2019-09-26:
> 哪个版本没有问题，大概日期？

这个真没注意。很少用这个路由器的wifi。之前以为是K3 漏油5G坏了。。用了一段时间没管。最近换了个op的原版固件，发现又好了。。然后重新刷回你的固件还是5G启动不了。~。。挂矿机还是你的好用啊~~=。=~~

ptpt52, 2019-09-26:
具体刷哪个版本的openwrt可以？固件路径发来看看

xztxy, 2019-09-26:
> 具体刷哪个版本的openwrt可以？固件路径发来看看

很早之前的固件，不知道怎么找路径什么的。不过我想起来了，上一次刷你的固件是在7月底的样子。那个时候的版本是没问题的。。。

xztxy, 2019-09-26:
> > 具体刷哪个版本的openwrt可以？固件路径发来看看
> 
> 很早之前的固件，不知道怎么找路径什么的。不过我想起来了，上一次刷你的固件是在7月底的样子。那个时候的版本是没问题的。。。
[image]

现在的版本刷完，弹出这个！

ptpt52, 2019-09-26:
正常的。

xztxy, 2019-09-26:
> 正常的。

刷完，重置了下，。。我改完信道又不行了。。~

ptpt52, 2019-09-26:
不改就行？

xztxy, 2019-09-26:
> 不改就行？

不好意思，没说清楚，刷完。不会出现之前的图中的未关联问题，但是实际还是不行的？wifi只能收到2G信号。

xztxy, 2019-09-30:
> 不改就行？

大佬。昨天更新的版本还是不行哦。。刷完2G 5G都没信号。。5G改不了信道。改了显示还是36、。并且依旧显示未关联~，麻烦大佬有空看下。谢谢！~~

ptpt52, 2019-09-30:
如果老版本没问题，建议用老版本代码编译打包试试
比如 https://github.com/x-wrt/x-wrt/tree/x-b201907270349

xztxy, 2019-09-30:
> 如果老版本没问题，建议用老版本代码编译打包试试
> 比如 https://github.com/x-wrt/x-wrt/tree/x-b201907270349

好的。，谢谢~！

ptpt52, 2020-03-15:
新版本也更新了k3驱动。

closed 2020-03-18 by ptpt52