x-wrt/x-wrt #52: 小米路由器3, wifi信号稳定问题

Guxing0321, 2019-10-16:
Hi Team,

使用了一段时间之后，发现信号的稳定性是个大问题，
经常出现2.4g 信号消失，只能用5G的wifi信号，然后，也撑不了太久，就没了.

之前是用老毛子的时候，没有出现，可以排除硬件的问题

型号： Xiaomi Mi Router R3
固件： x-wrt-5.0-b201910120941-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin

ptpt52, 2019-10-16:
最近的mtk无线确实有点问题，等过段时间看有没有修复吧

rommelniebres, 2020-09-25:
Hello I'm a beginner on this matter I'll try to explain my problem clearly. My Xiaomi Router 3 is also having a stability issue, after some hours( indefinite time) WIFI will crash and cannot be seen on the mobile phones WIFI settings, LAN connection is connected but with no internet (cannot also access 192.168.15.1) and i need to restart the router to gain access to the internet. I am using the latest build (-WRT 8.0_b202009190231 ). Please Help