x-wrt/x-wrt #60: acrh17编译报错

daiaji, 2019-12-04:
```
make[5]: Entering directory '/home/daiaji/x-wrt/build_dir/host/e2fsprogs-1.45.4/intl'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/daiaji/x-wrt/build_dir/host/e2fsprogs-1.45.4/intl'
making install in e2fsck
make[5]: Entering directory '/home/daiaji/x-wrt/build_dir/host/e2fsprogs-1.45.4/e2fsck'
        MKDIR_P /home/daiaji/x-wrt/staging_dir/host/bin /home/daiaji/x-wrt/staging_dir/host/share/man/man8
make[5]: config/install-sh: Command not found
make[5]: *** [Makefile:523: installdirs] Error 127
make[5]: Leaving directory '/home/daiaji/x-wrt/build_dir/host/e2fsprogs-1.45.4/e2fsck'
make[4]: *** [Makefile:434: install-progs-recursive] Error 1
make[4]: Leaving directory '/home/daiaji/x-wrt/build_dir/host/e2fsprogs-1.45.4'
make[3]: *** [Makefile:53: /home/daiaji/x-wrt/staging_dir/host/stamp/.e2fsprogs_installed] Error 2
make[3]: Leaving directory '/home/daiaji/x-wrt/tools/e2fsprogs'
time: tools/e2fsprogs/compile#0.15#0.05#0.19
make[2]: *** [tools/Makefile:161: tools/e2fsprogs/compile] Error 2
make[2]: Leaving directory '/home/daiaji/x-wrt'
make[1]: *** [tools/Makefile:157: /home/daiaji/x-wrt/staging_dir/host/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynnynynnyyynnyyy] Error 2
make[1]: Leaving directory '/home/daiaji/x-wrt'
make: *** [/home/daiaji/x-wrt/include/toplevel.mk:227：world] 错误 2
```
怎么整？

ptpt52, 2019-12-11:
可能编译环境不行呀，看编译文档

https://blog.x-wrt.com/docs/build/

daiaji, 2019-12-19:
@ptpt52 应该不是
这个问题复现起来就是你第一次编译开超多线程，比如24个，
这里就会报错……
单线程就不会
但这也太杀时间了……

daiaji, 2019-12-28:
似乎是网的问题，调整代理之后，能编译了

ptpt52, 2019-12-28:
编译没问题的，我们云自动化编译都通过了

closed 2020-03-08 by ptpt52