x-wrt/x-wrt #61: HC5761A USB网卡不可用，SD卡不可用

xpalm, 2019-12-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

ptpt52, 2019-12-11:
usb 网卡意思是 无线网卡吗？这个确实不支持，你需要可以编译相应的驱动。

sd卡我也没有设备测试过，无法验证

AmadeusGhost, 2020-03-26:
关于sd卡，LGA1150大佬那边有几个相关的commit：
https://github.com/LGA1150/openwrt/commit/0686f8a89275f3a481268a4fbfe19ed27d48f783 https://github.com/LGA1150/openwrt/commit/8b248d08563dae4e344adf5ec6dee459e3c45bd8 https://github.com/LGA1150/openwrt/commit/f456d02c893acdf4b3fcf0f58691611c5c83e70d
你可以自己测试一下

ptpt52, 2020-03-26:
好的，我合并这个几个commit 试试

easonjim, 2020-11-13:
经过测试，HC5761A最新版本的也是不能支持SD卡挂载