x-wrt/x-wrt #62: Xiaomi R3 6.0_b201912080827 kern.err kernel mt76x2e MCU message 31 (seq 14) timed out

adamenkow, 2019-12-09:
Now my Xiaomi R3 is on the latest version - **X-WRT 6.0_b201912080827 Eoan / LuCI branch git-19.340.10447-790eb98**

After flashing, I saw that WiFi is not active at all and the list of wireless interfaces it was designated as **Generic**. 
After the reboot, the interfaces went up often get driver related log entries
[image]

Now I get frequent errors in the logs:

Sun Dec  8 23:35:36 2019 kern.err kernel: [ 4740.685351] mt76x2e 0000:01:00.0: MCU message 31 (seq 14) timed out
Sun Dec  8 23:35:36 2019 kern.info kernel: [ 4740.709506] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Sun Dec  8 23:35:36 2019 kern.info kernel: [ 4740.715121] mt76x2e 0000:01:00.0: Build: 1
Sun Dec  8 23:35:36 2019 kern.info kernel: [ 4740.719370] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Sun Dec  8 23:35:36 2019 kern.info kernel: [ 4740.735719] mt76x2e 0000:01:00.0: Firmware running!
Sun Dec  8 23:35:36 2019 kern.info kernel: [ 4740.741394] ieee80211 phy0: Hardware restart was requested

I also tried to build (6.0_b201912080827) my firmware with minor changes - I got the same problem.
What to do with such a problem?

ptpt52, 2019-12-09:
it works fine, not error.

adamenkow, 2019-12-09:
> it works fine, not error.
Thank you for your quick response. Do I understand correctly that this is not worth paying attention to? I was worried about WLAN status after flashing "Generic"

closed 2020-04-04 by ptpt52