x-wrt/x-wrt #69: 在X-WRT下竞斗云怎么刷BL？

daiaji, 2019-12-19:
https://ericclose.github.io/R619AC-OpBoot-and-OpenWrt-flashing.html提到的`mtd -r write /tmp/opboot-p2w-r619ac-flash-v1.0.9.bin Bootloader`刷不上

ptpt52, 2019-12-20:
解锁版本的才能

daiaji, 2019-12-20:
@ptpt52  啥意思？
我用链接里的openwrt就能刷BL
但是x-wrt不能用`mtd -r write /tmp/opboot-p2w-r619ac-flash-v1.0.9.bin Bootloader`

ptpt52, 2019-12-20:
@daiaji QQ群才有下载哈哈。

ptpt52, 2019-12-20:
你自己修改dts编译固件也可以。

daiaji, 2019-12-21:
@ptpt52  改成编译时能产生这个二进制包如何？

hellodong, 2019-12-26:
@ptpt52, 大佬 ipq4019 hw crypto 弄好了吗，我跑 ipsec 还是 很低的带宽，软件加解密倒是可以不过不超50M
还有个事加你QQ群老是没反应

daiaji, 2019-12-27:
@hellodong AES硬件加速的话……
ARMv7是没有的 ARMv8才加入……

hellodong, 2019-12-27:
@daiaji , 这跟arm核没关系，是ipq40xx的外设 硬件加密引擎
在openwrt 仓库下老外专门在修复(ipq40xx: Qualcomm Crypto Engine fixes #2518),不过没啥进展
chacha20poly3015 这两天我也移上去了，ipsec 差不多能跑到70M。如果硬件加密引擎OK的话应该能跑到100M以上

daiaji, 2019-12-28:
@hellodong 有趣，我一直以为ipq40xx没有AES硬件加速，不过这个东西社区就算修不了，高通自己也会修吧……

daiaji, 2019-12-30:
@hellodong 那个能加速chacha20？不是应该只能加密AES吗？

hellodong, 2019-12-31:
选项里还没有，需要内核配置支持

ch1993128, 2020-02-03:
要ttl硬刷，被锁了的