x-wrt/x-wrt #79: Xiaomi R3 2.4ghz slow

erenkan, 2020-01-02:
5ghz working without problem but 2.4ghz extremly slow (between 1-5 mbps)

**Screenshots**

[image]

iRet, 2020-01-27:
2.4ghz works as it should for me on R3.

Romandas2, 2020-01-27:
> 2.4ghz works as it should for me on R3.

Are you sure? What did you do it to work?

erenkan, 2020-01-28:
> 2.4ghz works as it should for me on R3.

erenkan, 2020-01-28:
> 2.4ghz works as it should for me on R3.

Which X-wrt version are you using on R3?

Rast1234, 2020-01-28:
I tried x-wrt on my R3; it had generic driver for 2.4ghz. This driver does not work properly and causes slow speeds, client disconnects and loss of 2.4 network until reboot. Settings can't help.
Maybe some builds of x-wrt use ralink driver as pandorabox does?

iRet, 2020-01-29:
I did nothing special, just set my channel and band width to what I used with previous firmware on this devise. (3ch, 40mhz)

[image]

My client has internal antennas and ~15 meters far from the router.

[image]

arxarios, 2022-01-09:
[image]
I also get  slow wifi speeds at about 2Mbps and my providers connection is 100Mbps.
Is this bug? Is it going to be fixed?  Do i have to change some x-wrt settings to make the wifi work? Can you provide a guide to return to stock firmware ?