x-wrt/x-wrt #81: 设置自定义的app时报错

Mokee-nmy, 2020-01-14:
/usr/lib/lua/luci/model/cbi/passwall/node_config.lua:7: attempt to index field 'cbi' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/passwall/node_config.lua:7: in function 'get_customed_path'
	/usr/lib/lua/luci/model/cbi/passwall/node_config.lua:11: in function 'is_finded'
	/usr/lib/lua/luci/model/cbi/passwall/node_config.lua:63: in function 'e'
	/usr/lib/lua/luci/cbi.lua:48: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1013: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:766: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:378: in function </usr/lib/lua/luci/dispatcher.lua:377>

ptpt52, 2020-01-14:
注意这个是基于最新openwrt源码基础的系统，和你的app老版本可能存在不兼容。

Mokee-nmy, 2020-01-14:
> 注意这个是基于最新openwrt源码基础的系统，和你的app老版本可能存在不兼容。

已替换改app，新编译的APP安装后正常。

closed 2020-01-14 by Mokee-nmy