x-wrt/x-wrt #82: 可以加入 Ikve2服务支持 吗？

lwk523, 2020-01-19:
你好，  可以加入Ikve2 在 上iphone 有省电的效果呢 。 

谢谢了 。

ptpt52, 2020-01-19:
不支持。这玩意服务器太过复杂，而且体积不了，不适合几百个设备同时支持。

lwk523, 2020-01-22:
就是呢， Ikve2 太过复杂了。  那至于Softether 很多的WRT 固件都有支持 Softether服务器， 但就是没有一个有支持Softether的客端 。 很多人都误解了 Softether ，当它是 PPTP ，L2TP，和 Openvpn 而已， 而它还有 这个 Ethernet over HTTPS 的 protocol .   那这个可以加入吗 ? Softether的客端 .. 这很稳定的呢。  @ptpt52 

https://www.softether.org/@api/deki/files/12/=1.3.jpg

daiaji, 2020-02-06:
@lwk523 结论就是那为什么不分开裝？
物联网设备ROM本身就小……
最适合的是内核原生的WireGuard

zhhww58, 2020-02-29:
我自己把softether送上去 一直很稳定 softether不能所有版本一起怂 要分开 vpnserver vpnbridge vpnclient 分开，我的64M的r1cl 都能用 zram压一下稳稳地

lwk523, 2020-02-29:
> 我自己把softether送上去 一直很稳定 softether不能所有版本一起怂 要分开 vpnserver vpnbridge vpnclient 分开，我的64M的r1cl 都能用 zram压一下稳稳地

太好了， 我很需要vpnclient , 你是如何送上去的 ？  我是用 RT-ACRH17 呢