x-wrt/x-wrt #84: natcapd is active even when disabled in UI

iRet, 2020-01-27:
Hi there,

I've noticed that natcapd service is active even when I disable it.
Looks like a bug to me, please correct me if I am wrong.

Thank you.

Model	Xiaomi Mi Router R3
Firmware Version	X-WRT 7.0_b202001120437 Eoan / LuCI Master git-20.011.74058-b3174e9

```
root@X-WRT:~# ps aux | grep natcap
root      3371  0.0  0.8   1344  1028 ?        S    02:54   0:09 /bin/sh /usr/sbin/natcapd start
root      3731  0.0  0.7   1344   908 ?        S    02:54   0:01 /bin/sh /usr/sbin/natcapd start
root      3732  0.0  0.7   1344   908 ?        S    02:54   0:02 /bin/sh /usr/sbin/natcapd start
root     16212  0.0  0.6   1264   820 ?        S    13:31   0:00 cat /tmp/trigger_natcapd_update.fifo
root     16214  0.0  0.5   1264   644 ?        Ss   13:31   0:00 timeout 30 cat /tmp/trigger_natcapd_update.fifo
```        

[image]

ptpt52, 2020-01-27:
Natcap contains a variety of composite functions, including remote interface, proxy, fast forwarding natflow, will not be closed. None of these functions will work if it must be turned off. Run /etc/init.d/natcapd disable

iRet, 2020-01-29:
`/etc/init.d/natcapd disable` doesn't make any difference running processes wise. 
Is there any way to completely remove natcap packages? 
I remember you mentioned this on the lede forum back in the past. 
Thank you.

ptpt52, 2020-01-29:
reboot

closed 2020-01-30 by iRet

dangtrungluong, 2021-04-03:
opkg --force-removal-of-dependent-packages remove natcapd

vespersio, 2023-02-21:
> opkg --force-removal-of-dependent-packages remove natcapd
when run `/etc/init.d/firewall restart`
_sh: can't open '/usr/share/natcapd/firewall.include': No such file or directory_