x-wrt/x-wrt #86: 远程登录总是提示密码错误

hpp0hpp, 2020-02-11:
我用root/admin 用户名密码就登陆不上去，后来以为默认的无法登陆，改了发现还是提示密码错误

ptpt52, 2020-02-11:
莫非你的mac地址和别人的mac地址冲突，你实际登录了别人的机器？

ptpt52, 2020-02-11:
登录界面版本号，对得上？

hpp0hpp, 2020-02-11:
是一样的
[image]
[image]

ptpt52, 2020-02-11:
7.0 vs 6.0 版本。不一样。。。。

hpp0hpp, 2020-02-11:
咦，刚刚怎么看着一样，那我看看能不能改下mac

hpp0hpp, 2020-02-11:
[image]

[image]

在我点了去登陆后，就变成了6.0，不知道怎么回事

hpp0hpp, 2020-02-11:
换了个浏览器，进去没有去登陆，直接就是6.0 这是跟别人的mac 冲突了？

ptpt52, 2020-02-11:
就是你的mac不行，你拷贝了别人的eeprom吧。。。设备有问题，大量复制的垃圾设备。