x-wrt/x-wrt #89: HC5861 mt7620 2.4g wifi 冲突，死机

SuperWangCC, 2020-02-24:
**Describe the bug**
最新固件，5gwifi正常，而2.4g无法连接（一有客户端连接就会重启）

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

ptpt52, 2020-02-25:
你好，版本号发出来一下。

SuperWangCC, 2020-02-26:
b202002230943这个

closed 2020-05-02 by ptpt52