x-wrt/x-wrt #90: 建议增加对ACRH17的内核优化

ykzj, 2020-03-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
No.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
增加对ACRH的编译优化，如nat offload, cpu的neon和vfp指令优化。
或者提供.config文件供大家参考，我自行编译的文件没法引导，路由器反复重启
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

发现没法make kernel_menuconfig

ptpt52, 2020-03-09:
https://blog.x-wrt.com/docs/build/

ykzj, 2020-03-10:
按照文档，用.config模版进行删减后编译，还是没法引导。由于没有ttl线，看不到引导信息。能否给一些提示，哪些模块一定要选？