x-wrt/x-wrt #99: DDNS的一些东西希望验证一下……

daiaji, 2020-04-15:
现在的
/etc/hotplug.d/iface/95-ddns
```
#!/bin/sh
case "$ACTION" in
ifup)
/etc/init.d/ddns enabled && /usr/lib/ddns/dynamic_dns_updater.sh -n "$INTERFACE" -- start
;;
ifdown)
/usr/lib/ddns/dynamic_dns_updater.sh -n "$INTERFACE" -- stop
;;
esac
```
应该是没法用的……
假如我是从pppoe-wan获取IP地址
那是实际上触发的时候是执行了`/usr/lib/ddns/dynamic_dns_updater.sh -n "wan" -- start`
这不会执行成功
得用`/etc/init.d/ddns enabled && /usr/lib/ddns/dynamic_dns_updater.sh -n "$DEVICE" -- start`
然后DEVICE变量为pppoe-wan，这OK……
我在用cloudflare.com-v4……

ptpt52, 2020-04-15:
我靠，ddns竟然有这个bug

ptpt52, 2020-04-15:
借个 cloudflare.com 配置给我测试下？

daiaji, 2020-04-16:
@ptpt52  你可以用cloudflare.com的，免费的cdn，兼容你现在用的le证书，高级tls特性也都有，毕竟世界第一cdn，把ns转过去就OK了。github有私信吗？

daiaji, 2020-04-17:
@ptpt52 其他DDNS没问题？
总之从接口获取IP的话
cf-v4没法用……
但我检查源码提交
update_cloudflare_com_v4.sh和ddns.hotplug的修改都是有几个月了
不明白为什么会有这种问题……
内核的缘故？

ptpt52, 2020-04-20:
@daiaji  应该是没有问题的。

ptpt52, 2020-04-20:
直接说你实际遇到的问题还好些

daiaji, 2020-05-08:
> 直接说你实际遇到的问题还好些

就压根就没更新DDNS