x-wrt/x-wrt #105: 关于x-wrt的一些疑问

chenxudong2020, 2020-05-21:
1.x-wrt 内核比较lean的新 但是驱动太老 能否优化下驱动，驱动可以参考lean的源码中的驱动
他的驱动是从第三方github导入的；
2.x-wrt怎么github action全自动编译某个机型固件，例如我想编译hc5962并进行定制，麻烦出个示例；
3.x-wrt建议添加 上网时间控制插件 还有定时wifi插件 这两个大多数有需求；
4. x-wrt能否添加 开发者模式 在此模式下可以安装软件包，可在luci系统中添加此选线 如果选中开启此模式的话 可以放出软件包的菜单 ；
5.提个bug 其实改动比较麻烦 就是在系统无线中修改了无线ssid和密码等信息，但是设置向导中还显示默认的，可以做成空的 用户必填 否则不使用此功能（无法点击保存或者保存应用按钮）

chenxudong2020, 2020-05-21:
目前我自己安装了 wifi计划插件 是兼容此ｌｕｃｉ的

456vv, 2020-05-21:
这个内核是个问题，很多软件都安装不上。

chenxudong2020, 2020-05-21:
不是内核问题 是他的界面webif不兼容 很多软件装上 都没界面

ptpt52, 2020-05-21:
1 无线驱动，跟随openwrt官方上游，因此一般不会额外更新版本。
2 action 构建 https://github.com/x-wrt/build-release 参考
3 上网时间控制插件 还有定时wifi插件 有没有靠谱的开源代码，路径哪里，可以评估一下。
4 暂不提供安装软件包，界面简洁点好。安装ipk是高级操作，完全可以ssh登录 opkg命令执行
5 设置向导 这些不是问题，都有考虑过，目前情况没问题：如果设置向导被修改，以设置向导为主导，只有修改的当时设置向导才会工作。