x-wrt/x-wrt #110: 这段代码是不是不对？我查了cpuinfo是74kc但是这里是24kc

csqmdx, 2020-06-19:
_src/gz x-wrt_core http://downloads.openwrt.org/snapshots/targets/ar71xx/generic/packages
src/gz x-wrt_base http://downloads.openwrt.org/snapshots/packages/mips_24kc/base
src/gz x-wrt_luci http://downloads.openwrt.org/snapshots/packages/mips_24kc/luci
src/gz x-wrt_packages http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages
src/gz x-wrt_routing http://downloads.openwrt.org/snapshots/packages/mips_24kc/routing
src/gz x-wrt_telephony http://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony_

以上代码通过/etc/opkg# cat distfeeds.conf得到

cpuinfo

system type             : Qualcomm Atheros QCA956X ver 1 rev 0
machine                 : Phicomm K2T A1/A2/A3 board
processor               : 0
cpu model               : MIPS 74Kc V5.0

csqmdx, 2020-06-19:
因为opkg安装一个应用安不上就查了一下。

ptpt52, 2020-06-19:
没错的。你第三方源就不保证了。