x-wrt/x-wrt #111: 新版x-wrt，7620启动居然要3分钟

cnmyn, 2020-06-30:
手机两台7620的机器，优酷L1和极路由1S(HC5661)分别刷了
x-wrt-8.0-b202006281500-ramips-mt7620-hiwifi_hc5661-squashfs-sysupgrade.bin
x-wrt-8.0-b202006281500-ramips-mt7620-youku_yk1-squashfs-sysupgrade.bin
通电后用秒表测时，居然要3分钟才能启动
另外upnp好像也不行用，不能自动映射

ptpt52, 2020-06-30:
upnp问题是啥，具体啥情况，我这边一切正常。

cnmyn, 2020-07-04:
> upnp问题是啥，具体啥情况，我这边一切正常。

经过测试 好像发现了问题所在
当WAN口为内网时，UPNP会映射不成功，如移动宽带/二级路由，用TP测试可以成功
当WAN口为外网时，UPNP秒映射，用带电信公网测试的
所以看能不能修正一下，移动宽带确实有这个需求，配合全锥型NAT，体验会提升不少

ptpt52, 2020-07-05:
miniupnpd 官方实现就是按照这个标准来的，如果没有公网地址，他认为UPNP映射是不会成功的，不过确实我们的环境可以尝试一下，毕竟我们的环境没用公网但是网络是全锥形NAT的情形比较多。

cnmyn, 2020-07-05:
> miniupnpd 官方实现就是按照这个标准来的，如果没有公网地址，他认为UPNP映射是不会成功的，不过确实我们的环境可以尝试一下，毕竟我们的环境没用公网但是网络是全锥形NAT的情形比较多。

咱们这可以修改吗？或者换个其他功能一样的？移动挂机用得到这个，刚需！这一点优化的话，应该能提升一些X-WRT用户量

ptpt52, 2020-07-05:
貌似可以尝试配置stun实现，修改 /etc/config/upnpd配置，增加
```
	option use_stun '1'
	option stun_host 'stun.stunprotocol.org'
	option stun_port '3478'
```

你可以尝试下，例子是 stun服务器，如果不行请选择合适的stun服务器。

cnmyn, 2020-07-05:
> 貌似可以尝试配置stun实现，修改 /etc/config/upnpd配置，增加
> 
> ```
> 	option use_stun '1'
> 	option stun_host 'stun.stunprotocol.org'
> 	option stun_port '3478'
> ```
> 
> 你可以尝试下，例子是 stun服务器，如果不行请选择合适的stun服务器。

尝试了一下，用了你上面的服务器，我还找了几个，也不行
您看您这边还有更好的解决办法吗，不行就只有转投高恪了，高恪留空用不到，比较吃资源，还是期待用上x-wrt吧
主要是我不会编译固件

ptpt52, 2020-07-05:
那暂不支持，毕竟这个不符合规范。