x-wrt/x-wrt #113: VPN Router

michaelfsantos, 2020-07-01:
Hi @ptpt52,

I have found a cool router based on OpenWRT (InvizBox 2) with interesting features (multiple networks, multiple VPN associated with multiple SSID, ADBlocker, Tor, etc. - https://www.invizbox.com/products/ib2.) and cool web interface (https://www.vpncompare.co.uk/invizbox-2-review-vpn/)
The complete firmware is open source and can be found here on github:
https://github.com/invizbox/invizbox-2
https://github.com/invizbox/invizbox-feed
Would it be possible to fork this firmware and adapt it to support the routers that are already supported by x-wrt?
That would be great if possible! :)
Thanks!

ptpt52, 2020-07-01:
hello, we provide vpn router firmware, if u need it contact us

gitnewcomer, 2020-07-02:
Hi @ptpt52,
if you mean the X-WRT, I already tried it with my Mi Router 3 and it did not work.
The firmware from InvizBox 2 has all the functions that I would need to use my VPN provider.
With this firmware I can connect at the same time with more than one VPN profile using IKEv2 and associate each VPN profile to its own SSID, so I can have multiple SSID's that connect to multiple VPN's. It also has its own SSID for using Tor and own SSID for local network and clear internet.
Just check here: https://support.invizbox.com/hc/en-us/articles/360025879774-How-do-I-reach-the-Administration-Interface-on-my-InvizBox-2

_Note: All of the networks on your InvizBox 2 use different IP addresses. If you are trying to access the Administration Interface using an IP address as in step 3 above but are not connected to the first InvizBox 2 VPN network you will need to use one of the following addresses:

http://10.154.1.1 - for the 2nd VPN network
http://10.154.2.1 - for the 3rd VPN network
http://10.154.3.1 - for the 4th VPN network
http://10.154.4.1 - for the Tor network
http://10.154.5.1 - for the 1st Clear network
http://10.154.6.1 - for the 2nd Clear network
http://10.154.7.1 - for the Local network_

Is this possible on your VPN firmware?