x-wrt/x-wrt #116: Xiaomi Mi Router R3 - QoS over Nftable,s NFT-QoS - Not working

hayden-t, 2020-07-14:
Ive tried static and dynamic limits, and neither seem to work yet sqm qos works
X-WRT 8.0_b202007130713 Focal / LuCI Master git-20.192.51623-302decf

not sure how to debug, if just me or is broken ... thanks

ptpt52, 2020-07-14:
@hayden-t  you have to disable natflow, `System->Advanced Options`

hayden-t, 2020-07-14:
For me I only had 2 options in there, "full cone nat" and "enable fast forwarding" 
Disabling "full cone nat" worked ! thank, I would have never worked that out. :)

ptpt52, 2020-07-14:
disable fast forwarding please

hayden-t, 2020-07-14:
so this maybe related to this issue, but dynamic rule limti is not working https://github.com/rosywrt/nft-qos/issues/6

ptpt52, 2020-07-14:
no, just disable fast forwarding and everthing should work

hayden-t, 2020-07-14:
still no effect (static work but not dynamic)