x-wrt/x-wrt #117: AdBlock for x-WRT

mhamzas, 2020-07-14:
It'll be great if we can add adblock feature like OPENWRT has

ptpt52, 2020-07-14:
you may install it via ssh opkg

mhamzas, 2020-07-14:
I did try but no option coming in panel

#Configuring simple-adblock.
#simple-adblock is currently disabled.
#Run the following commands before starting service again:
#uci set simple-adblock.config.enabled='1'; uci commit simple-adblock;
#Configuring luci-app-simple-adblock.
#uci: Entry not found

mhamzas, 2020-07-14:
Actually I can access using this URL 
http://192.168.15.1/cgi-bin/luci/admin/services/simple-adblock

but not not adblock is not able to start

ptpt52, 2020-07-14:
`opkg install luci-app-adblock`

ptpt52, 2020-07-14:
logout and login again?

http://192.168.15.1/cgi-bin/luci/admin/services/adblock

my install log:
```
root@X-WRT:~# opkg update
Downloading http://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/freifunk/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_freifunk
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/freifunk/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_telephony
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.

root@X-WRT:~# opkg install luci-app-adblock
Installing luci-app-adblock (git-20.195.18875-4001402) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/luci-app-adblock_git-20.195.18875-4001402_all.ipk
Installing coreutils (8.32-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/coreutils_8.32-1_mipsel_24kc.ipk
Installing coreutils-sort (8.32-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/coreutils-sort_8.32-1_mipsel_24kc.ipk
Installing ca-bundle (20200601-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/ca-bundle_20200601-1_all.ipk
Installing adblock (4.0.6-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/adblock_4.0.6-1_all.ipk
Configuring coreutils.
Configuring coreutils-sort.
Configuring ca-bundle.
Configuring adblock.
Configuring luci-app-adblock.
```

mhamzas, 2020-07-14:
Thanks, it worked.
Can you also share the list of plugins supported by x-wrt ?

ptpt52, 2020-07-14:
every plugin supported by openwrt would work on x-wrt

closed 2020-07-14 by mhamzas