x-wrt/x-wrt #118: ARRIS SBR-AC1750 Bootloop

lrussell887, 2020-07-16:
I followed the instructions from [this post](https://www.right.com.cn/FORUM/thread-408165-2-1.html) to flash X-WRT onto my ARRIS SBR-AC1750 router. I downloaded the latest firmware files from the [release page](https://downloads.x-wrt.com/rom/), using files `x-wrt-8.0-b202007130713-ar71xx-nand-arris-sbr-ac1750-squashfs-kernel1.bin` and `x-wrt-8.0-b202007130713-ar71xx-nand-arris-sbr-ac1750-squashfs-rootfs1.bin`. I used the updated mtd command from [mleaf's website](https://www.mleaf.org/?id=9).

Here is my partition table I saved from the router before flashing:

```
dev:    size   erasesize  name
mtd0: 00100000 00020000 "u-boot"
mtd1: 00100000 00020000 "boot-flag"
mtd2: 00400000 00020000 "kernel"
mtd3: 02000000 00020000 "rootfs"
mtd4: 00400000 00020000 "kernel2"
mtd5: 02000000 00020000 "rootfs2"
mtd6: 01b00000 00020000 "config"
mtd7: 00400000 00020000 "openwrt"
mtd8: 00100000 00020000 "ft"
mtd9: 00200000 00020000 "PKI"
mtd10: 00100000 00020000 "caldata"
mtd11: 02400000 00020000 "kfs"
mtd12: 02400000 00020000 "kfs2"
mtd13: 00100000 00020000 "scfgmgr"
mtd14: 00100000 00020000 "uboot-rw"
```

Here is a recreation of all the commands I ran, since I do not have a log:

```
export PATH=/usr/sbin:/usr/bin:/sbin:/bin
cd /tmp

# This is how I backed up the old partitions

cat /dev/mtd2 > kernel.bin
cat /dev/mtd3 > rootfs.bin
cat /dev/mtd4 > kernel2.bin
cat /dev/mtd5 > rootfs2.bin

# I copied to my computer then deleted them

rm kernel.bin kernel2.bin rootfs.bin rootfs2.bin

# I copied files from local web server
# `kernel1.bin` is `x-wrt-8.0-b202007130713-ar71xx-nand-arris-sbr-ac1750-squashfs-kernel1.bin`
# `rootfs1.bin` is `x-wrt-8.0-b202007130713-ar71xx-nand-arris-sbr-ac1750-squashfs-rootfs1.bin`

wget http://192.168.0.2/libjson-c_0.12-1_ar71xx.ipk
wget http://192.168.0.2/libubox_2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb_ar71xx.ipk
wget http://192.168.0.2/mtd_21_ar71xx.ipk
wget http://192.168.0.2/kernel1.bin
wget http://192.168.0.2/rootfs.bin

# I installed the new mtd from mleaf's tutorial

opkg install libjson-c_0.12-1_ar71xx.ipk
opkg install libubox_2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb_ar71xx.ipk
opkg install mtd_21_ar71xx.ipk

# I checked MD5 and compared to original files, which was okay

md5sum kernel1.bin
md5sum rootfs1.bin

# I ran commands from ptpt52's tutorial

mtd write /tmp/kernel1.bin kernel
mtd write /tmp/kernel1.bin kernel2
mtd write /tmp/rootfs1.bin rootfs

# Then I rebooted

reboot
```

After I rebooted router it looks like it is stuck in a boot loop. There were no errors in the output of any of the steps. Do you know how I can fix it? All it shows are the flashing lights in the video, and my computer shows Ethernet disconnecting and reconnecting every few seconds. Is it possible I missed a step? I had to translate since I don't speak Chinese.

[Video of router bootloop](https://github.com/x-wrt/x-wrt/files/4928705/VID_20200715_204800.zip)

lrussell887, 2020-07-16:
I almost forgot -- I also saved a copy of `dmesg` I got from the stock firmware before flashing X-Wrt. Not sure if that has any more useful information.

[dmesg.txt](https://github.com/x-wrt/x-wrt/files/4929170/dmesg.txt)

ptpt52, 2020-07-16:
can't help, maybe the `kernel` partition is bad block