x-wrt/x-wrt #119: 关于HWNAT的疑问

MeIsReallyBa, 2020-08-03:
感谢加入HWNAT对无线的支持。在老毛子等用MTK SDK的固件里，有个参数FOE_TABLE_SIZE，使用的是16K，貌似是对应hwnat支持的最大连接数16384？现在在mtk_offload.h里相关的定义有这样几句，直接增大MTK_PPE_ENTRY_CNT的值可行否？

/* our table size is 4K */
#define MTK_PPE_ENTRY_CNT		0x1000
#define MTK_PPE_TBL_SZ			\
			(MTK_PPE_ENTRY_CNT * sizeof(struct mtk_foe_entry))

int mtk_ppe_debugfs_init(struct mtk_eth *eth);

ptpt52, 2020-10-18:
@MeIsReallyBa  可以修改的，但是同时要修改对应的 hash函数计算

MeIsReallyBa, 2020-10-22:
> @MeIsReallyBa 可以修改的，但是同时要修改对应的 hash函数计算

好的，看到最近NATFLOW也同步了dsa下的hwnat，请问natflow在dsa下实现了wifi也走hwnat吗？

ptpt52, 2020-10-23:
还不支持。

643406834, 2020-10-23:
@ptpt52 大神，liunx5.4 支持无线hwnat吗？

MeIsReallyBa, 2020-10-23:
> @ptpt52 大神，liunx5.4 无线支持hwnat吗？

这个和内核没关系，X-WRT的linux 5.4内核中，MTK的非dsa驱动已经能支持无线走HWNAT了（需要搭配NATFLOW)

ptpt52, 2021-02-28:
> > @MeIsReallyBa 可以修改的，但是同时要修改对应的 hash函数计算
> 
> 好的，看到最近NATFLOW也同步了dsa下的hwnat，请问natflow在dsa下实现了wifi也走hwnat吗？

现在支持了，不过如果开启两个gmac，第二个gmac做wan口的设备暂时不支持。

MeIsReallyBa, 2021-03-01:
> > ifi也走hwnat吗？
> 
> 现在支持了，不过如果开启两个gmac，第二个gmac做wan口的设备暂时不支持。

好的，UDP现在如何呢？

ptpt52, 2021-03-01:
udp 还是一样呀，一般支持，正常的udp连接，肯定双向都发很多包，所以没问题。
iperf3 发的udp就不是，正方向只发一个udp，所以没有到快速转发那一步

MeIsReallyBa, 2021-03-01:
> udp 还是一样呀，一般支持，正常的udp连接，肯定双向都发很多包，所以没问题。
> iperf3 发的udp就不是，正方向只发一个udp，所以没有到快速转发那一步

好的，还没测试。之前是pppoe的话udp多了一次就很难加速了。

MeIsReallyBa, 2021-04-17:
> udp 还是一样呀，一般支持，正常的udp连接，肯定双向都发很多包，所以没问题。
> iperf3 发的udp就不是，正方向只发一个udp，所以没有到快速转发那一步

另外请问下wan->5G(580M)比lan->5G(530M)快正常吗。。。

ptpt52, 2021-04-17:
正常

MeIsReallyBa, 2021-06-18:
> 正常

看见这个commit https://github.com/ptpt52/natflow/commit/203d1bb9a8e42f3783f3ad3cc08b6b7f3f4e2505
请问现在natflow支持gmac1了吗？按照这个https://github.com/x-wrt/x-wrt/commit/e7178d0103e0ee379a947a43b3ce64f5e6ad23bb 打开GMAC1，发现启用natflow后无线正常，有线不通。

ptpt52, 2021-06-18:
dsa 的 gmac1 支持的。
硬件上 gmac1 连接 port4 或者 port1

MeIsReallyBa, 2021-06-19:
> dsa 的 gmac1 支持的。
> 硬件上 gmac1 连接 port4 或者 port1

natflow现在支持gmac1了吗？现在启用gmac1后，不开natflow完全正常，开了natflow后无线正常，有线只能ping通wan口连接的设备。。。。

ptpt52, 2021-06-19:
支持的

MeIsReallyBa, 2021-06-19:
> 支持的

那难道是我姿势不对吗。。。启用natflow后，有线lan到wan不通，无线是通的。。。

ptpt52, 2021-06-19:
有些patch没同步吗

MeIsReallyBa, 2021-06-19:
> 有些patch没同步吗

我是最新的x-wrt。。。

MeIsReallyBa, 2021-06-19:
[  117.314351] debug: (PCO)[POST](natflow_path_post_ct_out_hook:1335)[192.168.1.228:56442->192.168.1.2:80|ID:6944,IL:52|......S.]: update pmtu from 0 to 1500
[  117.316952] debug: (PCO)[PRE](natflow_path_pre_ct_in_hook:750)[192.168.1.2:80->192.168.1.228:56442|ID:6CBD,IL:40|...A....]: pmtu=0 FRAG=00000000
[  117.316998] debug: (PCO)[POST](natflow_path_post_ct_out_hook:1335)[192.168.1.2:80->192.168.6.184:56442|ID:6CBD,IL:40|...A....]: update pmtu from 0 to 1500

日志看起来好像也是正常的，就是连不上，搞不懂了。。。

ptpt52, 2021-06-19:
晚点我确认下。

ptpt52, 2021-06-19:
@MeIsReallyBa  确认没有问题呀

client ---lan2(port2) ---wan(port4,gmac1)----server

MeIsReallyBa, 2021-06-19:
> @MeIsReallyBa 确认没有问题呀
> 
> client ---lan2(port2) ---wan(port4,gmac1)----server

太奇怪了，请问dts是https://github.com/x-wrt/x-wrt/commit/e7178d0103e0ee379a947a43b3ce64f5e6ad23bb 这样写的吧？

ptpt52, 2021-06-19:
没错呀就是这个设备。

MeIsReallyBa, 2021-06-19:
> 没错呀就是这个设备。

太奇怪了，那看来我这用不成gmac1了。。。开了natflow有线连不上，不开的话iperf3到wan又只有400M.。。。

ptpt52, 2021-06-19:
dts 看下？
另外 打开debug看看跑的时候日志 echo debug=7 >/dev/natflow_ctl

MeIsReallyBa, 2021-06-19:
[   93.033097] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:61740->192.168.1.2:5201|ID:26FD,IL:40|...A....]: dir=0 use hash=1034
[   93.046029] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:61740->192.168.1.2:5201|ID:26FD,IL:40|...A....]: dir=1 use hash=1926
[   93.059007] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:61740->192.168.1.2:5201|ID:26FE,IL:77|...AP...]: dir=1 use hash=1926
[   93.071956] info: (PCO) set hwnat offload 2 dev=lan1 s=192.168.1.2:5201 d=192.168.1.125:61740 dev=wan s=192.168.6.184:61740 d=192.168.1.2:5201
[   93.084921] info: do keep alive[1926]: 192.168.1.2:5201->192.168.1.125:61740 update jiffies=4294946600
[   93.094229] info: do keep alive update ct0: 192.168.1.2:5201->192.168.1.125:61740 diff_jiffies=1 HZ=100
[   93.335701] info: do keep alive[1034]: 192.168.6.184:61740->192.168.1.2:5201 update jiffies=4294946625
[   93.345035] info: do keep alive update ct1: 192.168.6.184:61740->192.168.1.2:5201 diff_jiffies=26 HZ=100
[   93.354536] info: do keep alive[1926]: 192.168.1.2:5201->192.168.1.125:61740 update jiffies=4294946625
[  100.648019] info: do keep alive[1926]: 192.168.1.2:5201->192.168.1.125:61740 update jiffies=4294947356
[  100.657329] info: do keep alive update ct0: 192.168.1.2:5201->192.168.1.125:61740 diff_jiffies=731 HZ=100
[  102.368718] info: do keep alive[1034]: 192.168.6.184:61740->192.168.1.2:5201 update jiffies=4294947528
[  102.378031] info: do keep alive update ct1: 192.168.6.184:61740->192.168.1.2:5201 diff_jiffies=903 HZ=100
[  102.387598] info: do keep alive[1926]: 192.168.1.2:5201->192.168.1.125:61740 update jiffies=4294947528
[  108.408058] info: do keep alive[1926]: 192.168.1.2:5201->192.168.1.125:61740 update jiffies=4294948132
[  108.417388] info: do keep alive update ct0: 192.168.1.2:5201->192.168.1.125:61740 diff_jiffies=604 HZ=100

 client:192.168.6.184(lan1)  <->router:192.168.1.125(gmac1 port4)<-> server:192.168.1.2

MeIsReallyBa, 2021-06-19:
/dts-v1/;
#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {	compatible = "xiaomi,mi-router-cr660x", "mediatek,mt7621-soc";
	model = "Xiaomi Mi Router CR660x";

	aliases {
		led-boot = &led_sys_yellow;
		led-failsafe = &led_sys_yellow;
		led-running = &led_sys_blue;
		led-upgrade = &led_sys_yellow;
		label-mac-device = &gmac0;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	leds {
		compatible = "gpio-leds";

		led_sys_yellow: sys_yellow {
			label = "yellow:sys";
			gpios = <&gpio 14 GPIO_ACTIVE_LOW>;
		};

		led_sys_blue: sys_blue {
			label = "blue:sys";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};

		net_yellow {
			label = "yellow:net";
			gpios = <&gpio 13 GPIO_ACTIVE_LOW>;
		};

		net_blue {
			label = "blue:net";
			gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
		};
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 18 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};
	};
};

&nand {
	status = "okay";

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		partition@0 {
			label = "Bootloader";
			reg = <0x0 0x80000>;
			read-only;
		};

		partition@80000 {
			label = "Nvram";
			reg = <0x80000 0x40000>;
			read-only;
		};

		partition@c0000 {
			label = "Bdata";
			reg = <0xc0000 0x40000>;
			read-only;
		};

		factory: partition@100000 {
			label = "Factory";
			reg = <0x100000 0x80000>;
			read-only;
		};

		partition@180000 {
			label = "crash";
			reg = <0x180000 0x40000>;
			read-only;
		};

		partition@1c0000 {
			label = "crash_log";
			reg = <0x1c0000 0x40000>;
			read-only;
		};

		/* "kernel" and "ubi" partition is the result of
		 * squashing next consecutive stock partitions:
		 * 1. firmware  0x0200000 - 0x2000000 (Stock firmware 0)
		 * 2. firmware1 0x2000000 - 0x3e00000 (Stock firmware 1)
		 * 3. overlay   0x3e00000 - 0x7000000 (Ubi overlay of stock fw)
		 * 4. obr       0x7000000 - 0x8000000 (Unallocated)
		 */

		partition@200000 {
			label = "kernel";
			reg = <0x200000 0x400000>;
		};

		partition@600000 {
			label = "ubi";
			reg = <0x600000 0x7980000>;
		};

		/*
		 * Leave 512 KiB for the bad block table
		 */
	};
};

&pcie {
	status = "okay";
};

&pcie0 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&ethernet {
	pinctrl-names = "default";
	pinctrl-0 = <&rgmii1_pins &rgmii2_pins &mdio_pins>;
};

&gmac0 {
	mtd-mac-address = <&factory 0x3fff4>;
};

&gmac1 {
	status = "okay";
	label = "wan";
	mtd-mac-address = <&factory 0x3fff4>;
	mtd-mac-address-increment = <(2)>;
	phy-mode = "rgmii-txid";
	phy-handle = <&phy4>;
};

&mdio {
	phy4: ethernet-phy@4 {
		reg = <4>;
	};
};

&switch0 {
	ports {
		port@0 {
			status = "okay";
			label = "lan1";
		};

		port@1 {
			status = "okay";
			label = "lan2";
		};

		port@2 {
			status = "okay";
			label = "lan3";
		};
	};
};

&state_default {
	gpio {
		groups = "i2c", "uart2", "uart3";
		function = "gpio";
	};
};

ptpt52, 2021-06-19:
natflow 版本不对吧？不是最新的？

MeIsReallyBa, 2021-06-19:
> natflow 版本不对吧？不是最新的？

natfllow是6-10的。。。。这是无线正常运行的log，对比了下比有线少了do keep alive update ct1，只有do keep alive update ct0:
[  863.026332] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:64464->192.168.1.2:5201|ID:8408,IL:40|...A....]: dir=0 use hash=448
[  863.039179] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:64464->192.168.1.2:5201|ID:8408,IL:40|...A....]: dir=1 use hash=2686
[  863.052162] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:64464->192.168.1.2:5201|ID:8409,IL:77|...AP...]: dir=1 use hash=2686
[  863.065093] info: (PCO) set hwnat offload 2 dev=wlan1 s=192.168.1.2:5201 d=192.168.1.125:64464 dev=wan s=192.168.6.161:64464 d=192.168.1.2:5201
[  863.078242] info: do keep alive[2686]: 192.168.1.2:5201->192.168.1.125:64464 update jiffies=56303
[  863.087105] info: do keep alive update ct0: 192.168.1.2:5201->192.168.1.125:64464 diff_jiffies=1 HZ=100
[  863.102159] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:64465->192.168.1.2:5201|ID:840D,IL:40|...A....]: dir=0 use hash=450
[  863.114997] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:64465->192.168.1.2:5201|ID:840D,IL:40|...A....]: dir=1 use hash=2172
[  863.128360] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:828)[192.168.1.125:64465->192.168.1.2:5201|ID:840E,IL:77|...AP...]: dir=1 use hash=2172
[  863.141336] info: (PCO) set hwnat offload 2 dev=wlan1 s=192.168.1.2:5201 d=192.168.1.125:64465 dev=wan s=192.168.6.161:64465 d=192.168.1.2:5201
[  863.154447] info: do keep alive[2172]: 192.168.1.2:5201->192.168.1.125:64465 update jiffies=56311
[  863.163354] info: do keep alive update ct0: 192.168.1.2:5201->192.168.1.125:64465 diff_jiffies=1 HZ=100

ptpt52, 2021-06-19:
不是，正常应该跑这个：
```
info: (PCO) set hwnat offload 1 dev=lan2 s=192.168.17.1:5201 d=192.168.18.223:56714 dev=wan s=192.168.15.101:56714 d=192.168.17.1:5201
```

你的事这样：
```
[ 863.065093] info: (PCO) set hwnat offload 2 dev=wlan1 s=192.168.1.2:5201 d=192.168.1.125:64464 dev=wan s=192.168.6.161:64464 d=192.168.1.2:5201
```

MeIsReallyBa, 2021-06-19:
那请问是什么问题呢。。natflow我看用的确实是最新的。。。

ptpt52, 2021-06-19:
kernel 的patch 同步最新吗？还是直接用 x-wrt 的仓库？

MeIsReallyBa, 2021-06-19:
直接用的xwrt的最新仓库

ptpt52, 2021-06-19:
```
diff --git a/natflow_path.c b/natflow_path.c
index 51207fb..0a472c4 100644
--- a/natflow_path.c
+++ b/natflow_path.c
@@ -954,6 +954,7 @@ fastnat_check:
                                                                                        }
 #endif
                                                                                        if (orig_dev->netdev_ops->ndo_flow_offload) {
+       printk("orig_dev=%s reply_dev=%s ndo_flow_offload=%p\n", orig_dev->name, reply_dev->name, reply_dev->netdev_ops->ndo_flow_offload);
                                                                                                if (orig_dev == reply_dev || reply_dev->netdev_ops->ndo_flow_offload) {
                                                                                                        struct natflow_offload *natflow = natflow_offload_alloc(ct, nf);
                                                                                                        struct flow_offload_hw_path orig = {

```

加一行打印调试看看

MeIsReallyBa, 2021-06-19:
> ```
> diff --git a/natflow_path.c b/natflow_path.c
> index 51207fb..0a472c4 100644
> --- a/natflow_path.c
> +++ b/natflow_path.c
> @@ -954,6 +954,7 @@ [...]

 orig_dev=wan reply_dev=lan1 ndo_flow_offload=00000000

不知道为啥这个patch打不上去，我还是手动去改的。。。。

ptpt52, 2021-06-19:
代码不一样？

MeIsReallyBa, 2021-06-19:
> 代码不一样？

github里的natflow和您电脑里的不一样吗？我后面直接fork的natflow再改的https://github.com/MeIsReallyBa/natflow/commit/95228069d73865d1514c6373c5570f7df4994746

ptpt52, 2021-06-19:
build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.124/drivers/net/ethernet/mediatek/mtk_eth_soc.c
```
2934         .set_rxnfc              = mtk_set_rxnfc,
2935 };
2936 
2937 static const struct net_device_ops mtk_netdev_ops = {
2938         .ndo_init               = mtk_init,
2939         .ndo_uninit             = mtk_uninit,
2940         .ndo_open               = mtk_open,
2941         .ndo_stop               = mtk_stop,
2942         .ndo_start_xmit         = mtk_start_xmit,
2943         .ndo_set_mac_address    = mtk_set_mac_address,
2944         .ndo_validate_addr      = eth_validate_addr,
2945         .ndo_do_ioctl           = mtk_do_ioctl,
2946         .ndo_tx_timeout         = mtk_tx_timeout,
2947         .ndo_get_stats64        = mtk_get_stats64,
2948         .ndo_fix_features       = mtk_fix_features,
2949         .ndo_set_features       = mtk_set_features,
2950 #ifdef CONFIG_NET_POLL_CONTROLLER
2951         .ndo_poll_controller    = mtk_poll_controller,
2952 #endif
2953         .ndo_flow_offload       = mtk_flow_offload,
2954 };
2955 
2956 static int mtk_add_mac(struct mtk_eth *eth, struct device_node *np)
2957 {
2958         const char *name = of_get_property(np, "label", NULL);
2959         const __be32 *_id = of_get_property(np, "reg", NULL);
2960         struct phylink *phylink;
2961         int phy_mode, id, err;
2962         struct mtk_mac *mac;
2963 
2964         if (!_id) {
2965                 dev_err(eth->dev, "missing mac id\n");
2966                 return -EINVAL;
2967         }
2968 
2969         id = be32_to_cpup(_id);
2970         if (id >= MTK_MAC_COUNT) {
2971                 dev_err(eth->dev, "%d is not a valid mac id\n", id);
2972                 return -EINVAL;
2973         }
2974 
2975         if (eth->netdev[id]) {
2976                 dev_err(eth->dev, "duplicate mac id found: %d\n", id);
2977                 return -EINVAL;
2978         }
```

这个路径的代码对得上？

ptpt52, 2021-06-19:
你打印出来，lan1 的 `ndo_flow_offload=00000000` 是 NULL ？

MeIsReallyBa, 2021-06-19:
这是我电脑里的，看了眼一样的吧
.set_rxnfc              = mtk_set_rxnfc,
};

static const struct net_device_ops mtk_netdev_ops = {
	.ndo_init		= mtk_init,
	.ndo_uninit		= mtk_uninit,
	.ndo_open		= mtk_open,
	.ndo_stop		= mtk_stop,
	.ndo_start_xmit		= mtk_start_xmit,
	.ndo_set_mac_address	= mtk_set_mac_address,
	.ndo_validate_addr	= eth_validate_addr,
	.ndo_do_ioctl		= mtk_do_ioctl,
	.ndo_tx_timeout		= mtk_tx_timeout,
	.ndo_get_stats64        = mtk_get_stats64,
	.ndo_fix_features	= mtk_fix_features,
	.ndo_set_features	= mtk_set_features,
#ifdef CONFIG_NET_POLL_CONTROLLER
	.ndo_poll_controller	= mtk_poll_controller,
#endif
	.ndo_flow_offload       = mtk_flow_offload,
};

static int mtk_add_mac(struct mtk_eth *eth, struct device_node *np)
{
	const char *name = of_get_property(np, "label", NULL);
	const __be32 *_id = of_get_property(np, "reg", NULL);
	struct phylink *phylink;
	int phy_mode, id, err;
	struct mtk_mac *mac;

	if (!_id) {
		dev_err(eth->dev, "missing mac id\n");
		return -EINVAL;
	}

	id = be32_to_cpup(_id);
	if (id >= MTK_MAC_COUNT) {
		dev_err(eth->dev, "%d is not a valid mac id\n", id);
		return -EINVAL;
	}

	if (eth->netdev[id]) {
		dev_err(eth->dev, "duplicate mac id found: %d\n", id);
		return -EINVAL;
	}

打印出来是这样的，C语言我也不太熟悉，应该不是null吧？
[image]

ptpt52, 2021-06-19:
你打出来是全000000

MeIsReallyBa, 2021-06-19:
所以现在看得出来问题在哪吗。。。。

ptpt52, 2021-06-19:
```
diff --git a/natflow_path.c b/natflow_path.c
index 51207fb..ac58fa3 100644
--- a/natflow_path.c
+++ b/natflow_path.c
@@ -954,6 +954,8 @@ fastnat_check:
                                                                                        }
 #endif
                                                                                        if (orig_dev->netdev_ops->ndo_flow_offload) {
+printk("orig_dev=%s reply_dev=%s ndo_flow_offload=%p ndo_flow_offload_check=%p\n",
+       orig_dev->name, reply_dev->name, reply_dev->netdev_ops->ndo_flow_offload, reply_dev->netdev_ops->ndo_flow_offload_check);
                                                                                                if (orig_dev == reply_dev || reply_dev->netdev_ops->ndo_flow_offload) {
                                                                                                        struct natflow_offload *natflow = natflow_offload_alloc(ct, nf);
                                                                                                        struct flow_offload_hw_path orig = {
```

你把。ndo_flow_offload_check 也打印一下看看

ptpt52, 2021-06-19:
确认是直接x-wrt仓库？这个文件没有缺少吧？
https://github.com/x-wrt/x-wrt/blob/master/target/linux/generic/pending-5.4/647-net-dsa-support-hardware-flow-table-offload.patch

MeIsReallyBa, 2021-06-19:
> 确认是直接x-wrt仓库？这个文件没有缺少吧？
> https://github.com/x-wrt/x-wrt/blob/master/target/linux/generic/pending-5.4/647-net-dsa-support-hardware-flow-table-offload.patch

[  161.136958] orig_dev=wan reply_dev=lan2 ndo_flow_offload=00000000 ndo_flow_offload_check=00000000
是的，直接x-wrt仓库。dsa的驱动如果不启用gmac1的话，开启natflow有线是正常的。

ptpt52, 2021-06-19:
你这个打印出来 ndo_flow_offload_check 也是 0000 说明很有问题了，这个地方打印 reply_dev=lan2 也是有问题，正常是 eth0

ptpt52, 2021-06-19:
dsa 那个 patch 你那里是不是没有？

ptpt52, 2021-06-19:
直接看kernel编译环境里 ，对应的文件，那个dsa patch是否打上了

MeIsReallyBa, 2021-06-19:
> dsa 那个 patch 你那里是不是没有？

647-net-dsa-support-hardware-flow-table-offload.patch 这个是有的，target下的东西我完全没动，我只替换了一些其他package，这个有影响吗。。。

ptpt52, 2021-06-19:
lan1 是 dsa，打印出的 ndo_flow_offload_check 应该是非空(000)的，不知道你的为什么不是。
肯定是内核代码不一样

ptpt52, 2021-06-19:
```
cat build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.124/.config | grep CONFIG_NF_FLOW_TABLE
# CONFIG_NF_FLOW_TABLE_INET is not set
CONFIG_NF_FLOW_TABLE=m
CONFIG_NF_FLOW_TABLE_HW=m
# CONFIG_NF_FLOW_TABLE_IPV4 is not set
# CONFIG_NF_FLOW_TABLE_IPV6 is not set
```

这个选项打开了吗？ CONFIG_NF_FLOW_TABLE

MeIsReallyBa, 2021-06-19:
> ```
> cat build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.124/.config | grep CONFIG_NF_FLOW_TABLE
> # CONFIG_NF_FLOW_TABLE_INET is not set
> CONFIG_NF_FLOW_TABLE=m [...]

感谢，发现问题了，因为怕和natflow冲突，我就没编译ipt-offload，跟着nf-flow也没编译进去，编译ipt-offload后正常了。。。。

MeIsReallyBa, 2021-06-21:
> ```
> cat build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.124/.config | grep CONFIG_NF_FLOW_TABLE
> # CONFIG_NF_FLOW_TABLE_INET is not set
> CONFIG_NF_FLOW_TABLE=m [...]

今天折腾了下原版的offload，发现这样稍微改下就可以开启ipv6的硬件加速，请问natflow可否考虑支持呢？

--- a/drivers/net/ethernet/mediatek/mtk_offload.c
+++ b/drivers/net/ethernet/mediatek/mtk_offload.c
@@ -46,6 +46,38 @@ mtk_offload_prepare_v4(struct mtk_eth *e
 	return 0;
 }
 
+static int
+mtk_offload_prepare_v6(struct mtk_eth *eth, struct mtk_foe_entry *entry,
+		       struct flow_offload_tuple *s_tuple,
+		       struct flow_offload_tuple *d_tuple,
+		       struct flow_offload_hw_path *src,
+		       struct flow_offload_hw_path *dest)
+{
+	int dest_port = 1;
+
+	if (dest->dev == eth->netdev[1])
+	    dest_port = 2;
+
+	mtk_foe_entry_prepare(entry, MTK_PPE_PKT_TYPE_IPV6_ROUTE_5T, s_tuple->l4proto,
+			      dest_port, dest->eth_src, dest->eth_dest);
+	mtk_foe_entry_set_ipv6_tuple(entry, 
+				     s_tuple->src_v6.s6_addr32, s_tuple->src_port,
+				     s_tuple->dst_v6.s6_addr32, s_tuple->dst_port);
+	
+				     
+				     
+
+	if (dest->flags & FLOW_OFFLOAD_PATH_PPPOE)
+		mtk_foe_entry_set_pppoe(entry, dest->pppoe_sid);
+
+	if (dest->flags & FLOW_OFFLOAD_PATH_VLAN)
+		mtk_foe_entry_set_vlan(entry, dest->vlan_id);
+
+	if (dest->flags & FLOW_OFFLOAD_PATH_DSA)
+		mtk_foe_entry_set_dsa(entry, dest->dsa_port);
+
+	return 0;
+}
 int mtk_flow_offload_add(struct mtk_eth *eth,
 			 enum flow_offload_type type,
 			 struct flow_offload *flow,
@@ -79,6 +111,11 @@ int mtk_flow_offload_add(struct mtk_eth
 		    mtk_offload_prepare_v4(eth, &reply, rtuple, otuple, dest, src))
 			return -EINVAL;
 		break;
+	case AF_INET6:
+		if (mtk_offload_prepare_v6(eth, &orig, otuple, rtuple, src, dest) ||
+		    mtk_offload_prepare_v6(eth, &reply, rtuple, otuple, dest, src))
+			return -EINVAL;
+		break;
 	default:
 		return -EINVAL;
 	}

ptpt52, 2021-06-21:
ipv6 nat环境不是很多，目前感觉不是很必要。目前还不想也没有空折腾。

MeIsReallyBa, 2021-06-21:
> ipv6 nat环境不是很多，目前感觉不是很必要。目前还不想也没有空折腾。

这个不是nat加速，是ipv6转发加速。。。。就家用pppoe拨号的情形

coolmaxter, 2021-12-03:
@MeIsReallyBa 你编译ipt-offload以后需要打开防火墙的硬件转发开关吗？ 我看似乎只需要依赖nf-flow

MeIsReallyBa, 2022-03-14:
> ipv6 nat环境不是很多，目前感觉不是很必要。目前还不想也没有空折腾。

请问7622的5.10 hwnat是我姿势不对吗，测速正常，但开启后tcp连接很容易建立失败。。。

ptpt52, 2022-03-14:
> > ipv6 nat环境不是很多，目前感觉不是很必要。目前还不想也没有空折腾。
> 
> 请问7622的5.10 hwnat是我姿势不对吗，测速正常，但开启后tcp连接很容易建立失败。。。

不知道你什么情况。echo debug=7 >/dev/natflow_ctl 开启调试打印信息看看情况。

MeIsReallyBa, 2022-04-12:
> > > ipv6 nat环境不是很多，目前感觉不是很必要。目前还不想也没有空折腾。
> > 
> > 
> > 请问7622的5.10 hwnat是我姿势不对吗，测速正常，但开启后tcp连接很容易建立失败。。。
> 
> 不知道你什么情况。echo debug=7 >/dev/natflow_ctl 开启调试打印信息看看情况。

昨天试了下，将dsa换成switch驱动，同时用最新版本的natflow，正常了。。。

ptpt52, 2022-05-28:
开始支持ipv6软件转发了

closed 2024-03-04 by ptpt52