x-wrt/x-wrt #121: 竞斗云交换机驱动问题

PussAzuki, 2020-08-19:
**Describe the bug**
A clear and concise description of what the bug is.

        自从7月3号开始我一个月没编译固件，然后按照之前的[混合配方](https://github.com/PussAzuki/Actions-OpenWrt) 发现wan/lan口工作不正常，无法正常与wan lan接着的设备交流，但是wan lan同时插着俩不同设备貌似竞斗云会直接变身为交换机；
        经过两周多时间的检查问题，发现你们的仓库下 /target/linux/ipq40xx/base-files/etc/board.d/02_network 这个文件下竞斗云的相关数据多了一点，由 "0t@eth0" "1:lan:4" "2:lan:3" "3:lan:2" "4:lan:1" "5:wan" 变为你们现在的 "0t@eth0" "1:lan:4" "2:lan:3" "3:lan:2" "4:lan:1" "5:wan:5" 而同一个文件内就竞斗云的变成了这样
        另外我用过 sed -i 's/192.168.1.1/192.168.7.1/g' package/base-files/files/bin/config_generate 的命令，op总不会是硬编码把这个给屏蔽了吧，我记得op官方也只动了ipq40xx的essedma交换机驱动
        希望你们能查明是否存在疏忽导致的问题。

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

ptpt52, 2020-08-20:
刷x-wrt发布的固件有问题吗？

PussAzuki, 2020-08-20:
0814版本是看起来没问题，后续的没试过；
按照你们提交的commit，我把与ipq40xx相关的全都手动改过了，但是就是开机后wan lan无反应；
但我把 ":5" 删除之后，然后就能工作了，现在也在用了

ptpt52, 2020-08-20:
反正没问题哦。跟:5无关。

closed 2022-04-21 by PussAzuki