x-wrt/x-wrt #122: MT7615DN (DBDC) - wireless config

kar200, 2020-08-26:
Hi guys,

Sorry I could not find a forum to as the question and I do not speak Chinese obviously. 

I have compiled a firmware for my own device (DIR-853). It uses the MT7615DN. I used openwrt and then found x-wrt and impressed with the HW-NAT. 

Anyway I saw that the only devices mentioned often is the phicomm k2p. It seems to be supported here as well. Luci does not have the correct details and I cannot only configure the router through (/etc/config/wireless).

My question is : does anyone have the K2P and using the wireless with it? Can you share please the wireless config file that works the best for you?

I managed to get both card working fine but they don't seem stable with different devices. Although I managed with the latest source from today to reach a speed of 650mbs with iperf3 (very impressive). But once I set it up in the kids area to be tested for real life scenario they tend to loose connection randomly.

Cheeers.

ptpt52, 2020-08-28:
hi, for mt7615 dbdc support we may get help from here https://github.com/openwrt/mt76/issues

As far as I know, dbdc is not fully supported for now.

kar200, 2020-08-28:
> hi, for mt7615 dbdc support we may get help from here https://github.com/openwrt/mt76/issues
> 
> As far as I know, dbdc is not fully supported for now.

Thanks a lot. I was going to post there but since it's a query about the config file and since I see a lot of K2P refrences here I thought I would get luckier here .

I will post the request there and see how it goes. DBDC does work for me (sort of) but with the last commits the boot process freezes (unless I set the interfaces to off ). This does not happen with X-wrt so I am guessing it has sth to do with dsa .

Thanks again for all the work you do for this frimware and for the reply .

K

closed 2020-08-29 by kar200