x-wrt/x-wrt #123: NFS Support

ZacharyFoxx, 2020-09-07:
Hello!

First of all, thank you very much for this alternative firmware that supports among other routers the Xiaomi Mi Router R3, which is what I have.

Well, is it possible to add NFS support to it? I need to enable this and tried to follow the OpenWRT tutorial at https://openwrt.org/docs/guide-user/services/nas/nfs.server, but the problem is that the necessary package, nfs-kernel-server, does not is available.

How to add support?

ptpt52, 2020-09-07:
hi, you could build your firmware from source, so that you could select `kmod-nfs` and `nfs-kernel-server` built-in

ZacharyFoxx, 2020-09-07:
Hello.

I appreciate your quick response.

Would you have the step by step how to do it?

ZacharyFoxx, 2020-09-08:
> hi, you could build your firmware from source, so that you could select `kmod-nfs` and `nfs-kernel-server` built-in

Hello,

I appreciate your quick response.

Is it not possible to just add support for NFS? I know it may sound silly, but I went looking for compiling OpenWRT (X-WRT is OpenWRT) and even on a topic in the OpenWRT forum people have difficulty in compiling the code: https://forum.openwrt.org/t/xiaomi-mi-wifi-3-support/2252

Please, please, just add support for NFS.

ptpt52, 2020-09-08:
simple steps to build:

use ubuntu 18.04

1.
```
sudo apt install build-essential ecj fastjar file flex g++ gawk gettext git git-core java-propose-classpath libelf-dev libncurses5-dev libncursesw5-dev libssl-dev python python3 subversion unzip wget zlib1g-dev

```

2.
```
git clone https://github.com/x-wrt/x-wrt.git
cd x-wrt
./scripts/feeds update -a
./scripts/feeds install -a

cp feeds/x/rom/lede/config.ramips-mt7620 .config

make menuconfig
#select kmod-nfs and nfs-kernel-server

make
```

ZacharyFoxx, 2020-09-08:
> simple steps to build:
> 
> use ubuntu 18.04
> 
> 
> ```
> sudo apt install build-essential ecj fastjar file flex g++ gawk gettext git git-core java-propose-classpath libelf-dev libncurses5-dev libncursesw5-dev libssl-dev python python3 subversion unzip wget zlib1g-dev
> ```
> 
> 
> ```
> git clone https://github.com/x-wrt/x-wrt.git
> cd x-wrt
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> 
> cp feeds/x/rom/lede/config.ramips-mt7620 .config
> 
> make menuconfig
> #select kmod-nfs and nfs-kernel-server
> 
> make
> ```

> simple steps to build:
> 
> use ubuntu 18.04
> 
> 
> ```
> sudo apt install build-essential ecj fastjar file flex g++ gawk gettext git git-core java-propose-classpath libelf-dev libncurses5-dev libncursesw5-dev libssl-dev python python3 subversion unzip wget zlib1g-dev
> ```
> 
> 
> ```
> git clone https://github.com/x-wrt/x-wrt.git
> cd x-wrt
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> 
> cp feeds/x/rom/lede/config.ramips-mt7620 .config
> 
> make menuconfig
> #select kmod-nfs and nfs-kernel-server
> 
> make
> ```

Hello,

Okay, I'll try.

I use Manjaro Linux. In my Slackware era, I even compiled a kernel, but that was a long time ago. I had even cloned the git repository and the rest of the steps, but I was really confused about the compilation options (I would even get the config.gz file in /proc/config.gz, if it was like a regular Linux).

Just to confirm, by doing this, using the file config.ramips-mt7620, I will have exactly as if it were a release of x-wrt (with the gui and the other common items of x-wrt), but with my modifications?

Oh, as I recall, I can manually edit the configuration file and:

- Comment the line CONFIG_TARGET_MULTI_PROFILE = y
- Uncomment the line # CONFIG_TARGET_ramips_mt7620_DEVICE_xiaomi_miwifi-r3 and leaving CONFIG_TARGET_ramips_mt7620_DEVICE_xiaomi_miwifi-r3 = y
- Comment on all TARGETS, except the one on xiaomi_miwifi-r3

This to save time in the compilation.

ptpt52, 2020-09-08:
don't Comment the line CONFIG_TARGET_MULTI_PROFILE = y

or you should select more options，say drivers and others

ZacharyFoxx, 2020-09-08:
> […](#)

Ok! I got it! Thank You =).

ZacharyFoxx, 2020-09-10:
> […](#)

Hello,

I did the procedures, it was enough to select "nfs-kernel-server" as a module, in the same way that it is chosen to support SAMBA (kmod-fs-ksmbd) that it automatically chose the dependencies, all packages were compiled, I upgraded using the sysupgrade image and apparently everything is fine ... but when I accessed via ssh and went to configure the NFS server, the packages were not installed together.

I checked the resulting .config file and I have this for xiaomi_miwifi-r3:

CONFIG_TARGET_DEVICE_PACKAGES_ramips_mt7620_DEVICE_xiaomi_miwifi-r3 = "aria2 blockd block-mount chat comgt comgt-ncm kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32c kmod-crypto-ctrod crypto-ctr km-crypto-ctr km kmod-crypto-ghash kmod-crypto-hmac kmod-crypto-md4 kmod-crypto-md5 kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha256 kmod-crypto-sha512 kmod-exfat-linux kmod-fs-autof4 kmod-fs-ext4 kmod-fs-ksmbd kmod-fs-msdos kmod-fs-vfat kmod-fuse kmod-gpio-button-hotplug kmod-lib-crc16 kmod-mii kmod-mt76x02-common kmod- mt76x2 kmod-mt76x2-common kmod-nls-cp437 kmod-rt2800-lib kmod-rt2800-mmio kmod-rt2800-soc kmod-rt2x00-lib kmod-rt2x00-mmio kmod-scsi-core kmod-usb2 kmod-usb-core kmod -usb-ehci kmod-usb-net kmod-usb-net-cdc-ether kmod-usb-net-cdc-mbim kmod-usb-net-cdc-ncm kmod-usb-net-huawei-cdc-ncm kmod-usb -net-ipheth kmod-usb-net-qmi-wwan kmod-usb-net-rndis kmod-usb-net-rtl8152 kmod-usb-net-sierrawireless kmod-usb-ohci kmod-usb-printer kmod-usb-serial km od-usb-serial-option kmod-usb-serial-qualcomm kmod-usb-serial-sierrawireless kmod-usb-serial-wwan kmod-usb-storage kmod-usb-storage-uas kmod-usb-wdm ksmbd-server libcares libnl -core libnl-genl libopenssl libopenssl-conf libpcre libsqlite3 libssh2 libstdcpp libusb-1.0 libxml2 luci-app-aria2 luci-app-ksmbd luci-app-openvpn luci-app-p910nd luci-app-wireguard luci-i18n -i18n-aria2-zh-cn luci-i18n-ksmbd-en luci-i18n-ksmbd-zh-cn luci-i18n-openvpn-en luci-i18n-openvpn-zh-cn luci-i18n-p910nd-en luci-i18n -p910nd-zh-cn luci-i18n-wireguard-en luci-i18n-wireguard-zh-cn luci-proto-3g luci-proto-ncm luci-proto-qmi luci-proto-wireguard luci-ssl-nginx nginx-mod -luci nginx-mod-luci-ssl nginx-ssl nginx-ssl-util nginx-util ntfs-3g openssl-util openvpn-openssl p910nd sendip uboot-envtools uqmi usb-modeswitch uwsgi uwsgi-cgi-plugin uwsgi-luci-support uwsgi-luci-support uwsgi -syslog-plugin webui-aria2 wget wireguard-tools wpad-openssl wsdd2 wwan "

The nfs-kernel-server_2.5.1-1_mipsel_24kc.ipk and nfs-utils-libs_2.5.1-1_mipsel_24kc.ipk packages were not included in the upgrade image.

How should I do to include them in the image?

ptpt52, 2020-09-10:
you `make menuconfig` and select them?

```
CONFIG_PACKAGE_nfs-kernel-server=y
```

ZacharyFoxx, 2020-09-10:
> you `make menuconfig` and select them?
> 
> ```
> CONFIG_PACKAGE_nfs-kernel-server=y
> ```

When you say "CONFIG_PACKAGE_nfs-kernel-server = y" it means embedding it, being marked with an "*", correct?
If so, I didn't select it like that, because on menuconfig the ksmbd-server package is marked "{M}", so I kept the same logic.

But now I understand, LOL. It was good because I learned how to have a local repository for opkg using a small flashdrive. Either way, supporting installing all packages, access via NFS worked perfectly.

Just one more question: Instead of cloning the git repository with "git clone https://github.com/x-wrt/x-wrt.git", I can use the zip releases available at https: // github. com / x-wrt / x-wrt / releases, or could that be a problem?

Thank You!

ptpt52, 2020-09-10:
>  I can use the zip releases available at https: // github. com / x-wrt / x-wrt / releases, or could that be a problem?

That would be more appropriate.

ZacharyFoxx, 2020-09-10:
> > I can use the zip releases available at https: // github. com / x-wrt / x-wrt / releases, or could that be a problem?
> 
> That would be more appropriate.

Do I use the release in zip format?

closed 2020-09-11 by ZacharyFoxx