x-wrt/x-wrt #129: ec2ns.ptpt52.com bombarding DNS

kolya182, 2020-10-22:
I have https://github.com/pi-hole/pi-hole as a local DNS server and found this log, @ptpt52 can you help to clarify ?
1. Why does make so many requests ?
2. What is the purpose of calling this url, can wrt works with out  ?
3. Is any way to change it to alternatives and how ?

ptpt52, 2020-10-22:
in package `natcapd` it is function of this PACKAGE
u could disable it by /etc/init.d/natcapd disable && reboot

or build your own firmware from source without `natcapd` built-in

kolya182, 2020-10-22:
> in package `natcapd` it is function of this PACKAGE
> u could disable it by /etc/init.d/natcapd disable && reboot
> 
> or build your own firmware from source without `natcapd` built-in

But does it do ?

ptpt52, 2020-10-22:
> > in package `natcapd` it is function of this PACKAGE
> > u could disable it by /etc/init.d/natcapd disable && reboot
> > or build your own firmware from source without `natcapd` built-in
> 
> But what does it do ?

https://github.com/ptpt52/natcap

ptpt52, 2020-10-23:
it is multi functions on natcapd
1. it provide a method to proxy network
2. it support on-key-vpn
3. it provide remote-mgr access via https url
4. it support for soft/hardware offload nat

ericwang2006, 2021-06-13:
> it is multi functions on natcapd
> 
> 1. it provide a method to proxy network
> 2. it support on-key-vpn
> 3. it provide remote-mgr access via https url
> 4. it support for soft/hardware offload nat

其实我只想要第4项功能，不知道有没有什么方法可以不启用natcapd，但是可以启用hardware offload nat？
我知道有可能是有特殊原因不方便把这几项功能分开。

remlei, 2023-06-04:
I also experience this issue but yeah, it is natcapd related, as long as you remove that package, you should not receiving that kind of DNS bombarding issue anymore.

its basically function like any html  based reverse proxy service out there like nginx, haproxy, cloudflare argotunnel, etc....