x-wrt/x-wrt #131: bug about mwan3 or maybe another

pythonwood, 2020-10-25:
```
# route -n   # wg is up , mwan3 is installed
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         x.4x.2x6.1     0.0.0.0         UG    30     0        0 pppoe-wan
10.11.0.0       0.0.0.0         255.255.255.0   U     20     0        0 wg1
x.4x.2x6.1      0.0.0.0         255.255.255.255 UH    0      0        0 pppoe-wan
xx.x.19.x5  x.4x.2x6.1     255.255.255.255 UGH   30     0        0 pppoe-wan
192.168x.0     0.0.0.0         255.255.255.0   U     0      0        0 br-lan
```

```
ping  10.11.0.1  # wg test is ok

ping -I wg1 10.11.0.1   # wg test is ok too

ping -I wg1  8.8.8.8    ##  hang on.  nohting output
PING 8.8.8.8 (8.8.8.8) from 10.11.0.4 wg1: 56(84) bytes of data.

opkg remove --autoremove --force-removal-of-dependent-packages   mwan3   # uninstall mwan3

ping -I wg1  8.8.8.8  # ok after uninstalling mwan3
64 bytes from 10.11.0.1: icmp_seq=1 ttl=64 time=204 ms

```

It will fail again if **opkg install mwan3**.

ptpt52, 2020-10-26:
yep , trying to fix it in next release.

pythonwood, 2020-10-29:
it take me lots of day to try to resolve but failed.

thanks a lot.

ptpt52, 2020-10-29:
fixed.

pythonwood, 2020-10-29:
em,  fix in which commit or branch ?

ptpt52, 2020-10-29:
the latest commit https://github.com/x-wrt/x-wrt/commit/2a2cc9b4b92583aec8606b00dc1c3229a46ea025