x-wrt/x-wrt #138: Xiaomi Mi Router R3 WDS not working

mohalisad, 2020-11-17:
**Describe the bug**
When I enable wds for a wireless client after that I cannot ping any ip in that subnet from router
Wds broke the wireless connection;

**Expected behavior**
Wds worked and the connection not broke

ptpt52, 2020-11-18:
@mohalisad 

Unable to reproduce your problem
wds works normally, did you forget to set the wds AP mode of the opposite end?

mohalisad, 2020-11-18:
Thanks for your response
The actual problem that I have is the original firmware mi router has a repeater feature and that worked with my current network fantasticly now I need to replicate that feature in x wrt without changing the ap side

how can i do that?

I mean what technology the original fw used to do that

ptpt52, 2020-11-18:
not really sure how the oem firmware do it
but if you want to get wds client work on openwrt, AP wds should be enable on the other end

mohalisad, 2020-11-18:
Ok
thanks in advance

closed 2020-11-18 by mohalisad