x-wrt/x-wrt #146: Problem with dhcp

M0H4MM4D0UTIS, 2020-12-22:
Hi
After rebooting the modem, the computer will not connect to the modem
The modem gives an incorrect IP (Does not follow Static Lease)
I have to disable and enable the network card once to fix it (Or disconnect and reconnect the cable)

This problem is only with the x-wrt firmware and not on other firmwares before (not a problem with my computer or Windows - I just installed the firmware)

Static Lease: 192.168.1.100 -> 04:92:26:d4:80:90

Model: Xiaomi Mi Router R3
Architecture: MediaTek MT7620A ver:2 eco:6
Firmware Version: X-WRT 8.0_b202012211725 Focal / LuCI Master git-20.355.32379-399c049
Kernel Version: 5.4.83

my systemlog after reboot:

> Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: Connected to system UBus
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: started, version 2.82 cachesize 2048
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: UBus support enabled: connected to system bus
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq-dhcp[4973]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain test
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain onion
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain localhost
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain local
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain invalid
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain bind
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain lan
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain test
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain onion
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain localhost
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain local
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain invalid
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain bind
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using only locally-known addresses for domain lan
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using nameserver 8.8.8.8#53
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: using nameserver 4.2.2.4#53
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: read /etc/hosts - 4 addresses
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: read /tmp/hosts/odhcpd - 1 addresses
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq[4973]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Dec 23 02:39:23 2020 daemon.info dnsmasq-dhcp[4973]: read /etc/ethers - 0 addresses
Wed Dec 23 02:39:23 2020 daemon.notice miniupnpd[5001]: private/reserved address checking is ignore
Wed Dec 23 02:39:23 2020 daemon.notice miniupnpd[5001]: HTTP listening on port 5000
Wed Dec 23 02:39:23 2020 daemon.notice miniupnpd[5001]: Listening for NAT-PMP/PCP traffic on port 5351
Wed Dec 23 02:39:23 2020 user.notice nft-qos-monitor: ACTION=update, MACADDR=04:92:26:d4:80:90, IPADDR=192.168.1.100, HOSTNAME=DESKTOP-8ASIN6V
Wed Dec 23 02:39:24 2020 kern.info kernel: [   69.324546] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Wed Dec 23 02:39:24 2020 kern.info kernel: [   69.331342] br-lan: port 3(wlan0) entered blocking state
Wed Dec 23 02:39:24 2020 kern.info kernel: [   69.336816] br-lan: port 3(wlan0) entered forwarding state
Wed Dec 23 02:39:24 2020 daemon.notice netifd: Network device 'wlan0' link is up
Wed Dec 23 02:39:24 2020 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Wed Dec 23 02:39:24 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Wed Dec 23 02:39:24 2020 user.notice nft-qos-dynamic: ACTION=update, MACADDR=04:92:26:d4:80:90, IPADDR=192.168.1.100, HOSTNAME=DESKTOP-8ASIN6V
Wed Dec 23 02:39:25 2020 user.notice nft-qos-monitor: ACTION=update, MACADDR=04:92:26:d4:80:90, IPADDR=192.168.1.100, HOSTNAME=DESKTOP-8ASIN6V
Wed Dec 23 02:39:25 2020 daemon.notice procd: /etc/rc.d/S99nft-qos: sh: dynamic: unknown operand
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Dec 23 02:39:25 2020 daemon.notice hostapd: ACS: Survey is missing noise floor
[... 149 lines cut ...]
Wed Dec 23 02:39:30 2020 daemon.err uwsgi[2913]: Wed Dec 23 02:39:30 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Wed Dec 23 02:39:30 2020 user.notice natcapd: Reloading cone_wan_ip due to ifup of loopback (lo)
Wed Dec 23 02:39:31 2020 daemon.info procd: - init complete -
Wed Dec 23 02:39:31 2020 user.info wsdd2[5712]: starting.
Wed Dec 23 02:39:31 2020 user.notice natcapd: Reloading natcap_connected due to ifup of loopback (lo)
Wed Dec 23 02:39:33 2020 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Wed Dec 23 02:39:33 2020 user.notice mwan3-hotplug[5968]: mwan3 hotplug on wan not called because interface disabled
Wed Dec 23 02:39:34 2020 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Wed Dec 23 02:39:36 2020 user.notice natcapd: Reloading cone_wan_ip due to ifup of wan (eth0.2)
Wed Dec 23 02:39:36 2020 user.notice natcapd: Reloading natcap_connected due to ifup of wan (eth0.2)
Wed Dec 23 02:39:37 2020 daemon.notice miniupnpd[5001]: shutting down MiniUPnPd
Wed Dec 23 02:39:37 2020 daemon.notice miniupnpd[6458]: private/reserved address checking is ignore
Wed Dec 23 02:39:37 2020 daemon.notice miniupnpd[6458]: HTTP listening on port 5000
Wed Dec 23 02:39:37 2020 daemon.notice miniupnpd[6458]: Listening for NAT-PMP/PCP traffic on port 5351
Wed Dec 23 02:40:32 2020 daemon.info uwsgi-luci: luci: accepted login on /admin/initsetup for root from 192.168.1.100
Wed Dec 23 02:40:33 2020 daemon.err nginx[3135]: 2020/12/23 02:40:33 [error] 3289#0: *4 "/www/ubus/index.html" is not found (2: No such file or directory), client: 192.168.1.100, server: , request: "POST /ubus/?1608678631422 HTTP/1.1", host: "192.168.1.1", referrer: "http://192.168.1.1/cgi-bin/luci/admin/initsetup"
Wed Dec 23 02:40:35 2020 daemon.info hostapd: wlan1: STA 30:ab:6a:3c:1e:54 IEEE 802.11: authenticated
Wed Dec 23 02:40:35 2020 daemon.info hostapd: wlan1: STA 30:ab:6a:3c:1e:54 IEEE 802.11: associated (aid 1)
Wed Dec 23 02:40:35 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 30:ab:6a:3c:1e:54
Wed Dec 23 02:40:35 2020 daemon.info hostapd: wlan1: STA 30:ab:6a:3c:1e:54 RADIUS: starting accounting session 27BC5DAA667657D0
Wed Dec 23 02:40:35 2020 daemon.info hostapd: wlan1: STA 30:ab:6a:3c:1e:54 WPA: pairwise key handshake completed (RSN)
Wed Dec 23 02:40:36 2020 daemon.err uwsgi[2913]: *** has_emperor mode detected (fd: 8) ***
Wed Dec 23 02:40:36 2020 daemon.err uwsgi[2913]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-cgi_io.ini
Wed Dec 23 02:40:36 2020 daemon.err uwsgi[2913]: Wed Dec 23 02:40:36 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini has been spawned
Wed Dec 23 02:40:36 2020 daemon.err uwsgi[2913]: Wed Dec 23 02:40:36 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini is ready to accept requests

ptpt52, 2020-12-24:
What topology ?

M0H4MM4D0UTIS, 2020-12-24:
> What topology ?
@ptpt52 
I use the star topology

M0H4MM4D0UTIS, 2020-12-29:
My problem was solved in the new firmware version (build 202012281647)
Thank you for your efforts

closed 2020-12-29 by M0H4MM4D0UTIS