x-wrt/x-wrt #147: 选择其它主题，Luci会崩溃

wavelet001, 2020-12-25:
make menuconfig加入luci-theme-freifunk-generic主题，编译后的固件选择它，Web界面就崩溃了（有出错提示，忘记截图了），其它主题暂时还没加入.config中，不知道是不是也有这个出现这种情况。

ptpt52, 2020-12-25:
并没有测试过这个主题，暂不知道。

wavelet001, 2020-12-25:
`/usr/lib/lua/luci/template.lua:97: Failed to execute template 'view'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/freifunk-generic/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/freifunk-gene..."]:20: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/freifunk-gene..."]:20: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:759: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:759: in function 'include'
	[string "/usr/lib/lua/luci/view/view.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:967: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:482: in function </usr/lib/lua/luci/dispatcher.lua:481>
[image]
`

wavelet001, 2020-12-25:
[image]

wavelet001, 2020-12-25:
我另外还测试了第三方luci-theme-argon主题[https://github.com/jerrykuku/luci-theme-argon](url)，也是这样的问题。

ptpt52, 2020-12-25:
其他主题不考虑，luci-theme-bootstrap luci-theme-openwrt-2020 是支持的。

closed 2020-12-29 by wavelet001