x-wrt/x-wrt #148: 新增软件失败

wavelet001, 2020-12-25:
我在.config中同时新增4个软件：
`CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-frps=y
CONFIG_PACKAGE_luci-app-hd-idle=y
CONFIG_PACKAGE_luci-app-smartdns=y`
结果编译出来的固件，没有网络，但是如果单独新增其中一个软件却一切正常。

[image]

[image]

wavelet001, 2020-12-25:
再次编译了一次，现在只加入：
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-frps=y
CONFIG_PACKAGE_luci-app-smartdns=y
故障依旧。

ptpt52, 2020-12-25:
加入这几个，固件大小是否超了？

.config 里面 CONFIG_TARGET_ROOTFS_PARTSIZE= 设置 root 分区大小

wavelet001, 2020-12-25:
再次编译了一次，这次这样的组合就一切正常了。
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-hd-idle=y
CONFIG_PACKAGE_luci-app-smartdns=y

wavelet001, 2020-12-25:
但是登陆页面却出现了SmartDNS的选项。

[image]

ptpt52, 2020-12-25:
有些app 太老了不兼容很正常。官方源才正常。

wavelet001, 2020-12-25:
请问默认配置用的是不是官方源？

ptpt52, 2020-12-25:
你浏览器缓存问题吧

wavelet001, 2020-12-25:
> 你浏览器缓存问题吧

不是，我用的是隐私模式，而且还使用了不同的浏览器。

wavelet001, 2020-12-25:
> 加入这几个，固件大小是否超了？
> 
> .config 里面 CONFIG_TARGET_ROOTFS_PARTSIZE= 设置 root 分区大小

大神就是大神，将它设置成1024就一切正常了。

ptpt52, 2020-12-25:
> > 加入这几个，固件大小是否超了？
> > .config 里面 CONFIG_TARGET_ROOTFS_PARTSIZE= 设置 root 分区大小
> 
> 大神就是大神，将它设置成1024就一切正常了。

也不是改越大越好，改刚好合适就行。
boot分区是 CONFIG_TARGET_KERNEL_PARTSIZE=
这两个分区，加起来，如果刚好256最好，如果256不够，就512，……

这两个分区加起来的大小，就是生产固件的大小（解压后）
影响刷机速度，影响刷机时候占用内存大小。

wavelet001, 2020-12-26:
> > > 加入这几个，固件大小是否超了？
> > > .config 里面 CONFIG_TARGET_ROOTFS_PARTSIZE= 设置 root 分区大小
> > 
> > 
> > 大神就是大神，将它设置成1024就一切正常了。
> 
> 也不是改越大越好，改刚好合适就行。
> boot分区是 CONFIG_TARGET_KERNEL_PARTSIZE=
> 这两个分区，加起来，如果刚好256最好，如果256不够，就512，……
> 
> 这两个分区加起来的大小，就是生产固件的大小（解压后）
> 影响刷机速度，影响刷机时候占用内存大小。

哦！原来是这样，谢谢，谢谢！

closed 2021-01-07 by wavelet001