x-wrt/x-wrt #150: 疑似ipq40xx系列TX/RX功能存在问题

PussAzuki, 2020-12-27:
**Describe the bug**
如题所示

**To Reproduce**
Steps to reproduce the behavior:
使用12.24版本的固件
在路由器上使用6in4提供的ipv6
然后使用AGH并设置ipv6 dns 服务器
tcpdump显示出现ipv6 tx cksum错误，路由器下的任意主机无此问题

**Additional context**
Add any other context about the problem here.
我怀疑是之前的实现 time-balanced scheduling 的
target/linux/ipq40xx/patches-5.4/715-essedma-refine-txq-to-be-adaptive-of-cpus-and-netdev.patch及 target/linux/ipq40xx/patches-5.4/716-essedma-reduce-write-reg.patch
的缺失导致的
更早之前跟着操作无此问题

我是以为op官方把内容也塞进去了，然后直接给主路由更新了，然后路由器上 stubby clash dcompass mosdns AGH都提示无法使用ipv6 dns 上游
但是看lean那边没删这两个补丁也没有相关issue 于是大胆猜测是你这的问题

ptpt52, 2020-12-28:
啥问题？确定是官网下载的固件有问题吗？我也用ipq40xx，没发现。

PussAzuki, 2020-12-28:
无法在路由器本身上使用类似AGH之类的软件来使用IPv6 DoT/DoH ，会显示网不通
tcpdump抓以那个IPv6地址为目标的包发现是 cksum incorrect ，谷歌搜到可能是因为offload/TCPMSS的问题

ptpt52, 2020-12-28:
确定是x-wrt官网下载的固件有问题吗？

PussAzuki, 2020-12-28:
x-wrt-8.0-b202012211725-ipq40xx-generic-p2w_r619ac-128m-squashfs-nand-factory.ubi
测试了这个 是有此问题的

PussAzuki, 2020-12-28:
我顺便也看了下QSDK公开部分，发现那里是有你补丁里的内容的

ptpt52, 2020-12-28:
试试更早的版本有没有问题呢 比如 https://github.com/x-wrt/build-release/releases/tag/v8.0_b202010281419

PussAzuki, 2020-12-28:
有上面两个补丁的时候是绝对能用的...我现在都不太敢试了，再断网一阵子可不好...

ptpt52, 2020-12-28:
跟那两个补丁关系不大，那两个是有的，只不过不是补丁形式，已经整合进代码了。

PussAzuki, 2020-12-28:
[查看可选的接口](https://i.loli.net/2020/12/28/fa8c1BUrAmNiqoJ.png)
[尝试使用IPv6的DoT](https://i.loli.net/2020/12/28/MLykYvK1Ec7NjmZ.png) 上面是路由器下的主机，下面是路由器本身
[路由器上对IPv6 DNS进行ping测试](https://i.loli.net/2020/12/28/sdBQEXkSm3hyACR.png)

如果有更多我能提供的东西的话我很乐意提供
我其实不确定是你这的问题还是op的防火墙有个设定改了的导致问题，希望能帮忙看看

ptpt52, 2020-12-28:
对比过代码，和之前的基本一样，所以可能是你自己编译的固件有问题吧，x-wrt发布的固件应该没问题。

PussAzuki, 2020-12-28:
```cp ../lga1150/package/firmware/ipq-wifi/board-p2w_r619ac.qca4019 package/firmware/ipq-wifi

mkdir package/network/config/firewall/patches

cp ../lga1150/package/network/config/firewall/patches/fullconenat.patch package/network/config/firewall/patches

cp -r ../lga1150/package/network/fullconenat package

cp ../lga1150/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-r619ac* target/linux/ipq40xx/files/arch/arm/boot/dts

cp ../lga1150/target/linux/generic/hack-5.4/952-net-conntrack-events-support-multiple-registrant.patch target/linux/generic/hack-5.4

mkdir target/linux/ipq40xx/base-files/etc/hotplug.d/net

cp ../x-wrt/target/linux/ipq40xx/base-files/etc/hotplug.d/net/21_adjust_network target/linux/ipq40xx/base-files/etc/hotplug.d/net

cp ../x-wrt/target/linux/ipq40xx/base-files/lib/adjust_network.sh target/linux/ipq40xx/base-files/lib

cp ../x-wrt/target/linux/ipq40xx/patches-5.4/901-essedma-disable-default-vlan-tagging.patch target/linux/ipq40xx/patches-5.4

cp ../lga1150/target/linux/ipq40xx/patches-5.4/999-cpufreq-ondemand-boost-freq.patch target/linux/ipq40xx/patches-5.4

cp ../x-wrt/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma

cp ../x-wrt/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.h target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma

cp ../x-wrt/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma_axi.c target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma

cp ../x-wrt/target/linux/ipq40xx/files/drivers/net/phy/ar40xx.c target/linux/ipq40xx/files/drivers/net/phy

这是直接复制的部分

diff --git a/target/linux/ipq40xx/base-files/etc/board.d/02_network b/target/linux/ipq40xx/base-files/etc/board.d/02_network
index fe21dc8035..f6d648c56b 100755
--- a/target/linux/ipq40xx/base-files/etc/board.d/02_network
+++ b/target/linux/ipq40xx/base-files/etc/board.d/02_network
@@ -115,6 +115,11 @@ ipq40xx_setup_interfaces()
 		ucidef_add_switch "switch0" \
 			"0u@eth0" "1:lan" "2:lan" "3:lan" "4:lan" "0u@eth1" "5:wan"
 		;;
+	p2w,r619ac*|)
+		ucidef_set_interfaces_lan_wan "eth1" "eth0"
+		ucidef_add_switch "switch0" \
+			"0t@eth0" "1:lan:4" "2:lan:3" "3:lan:2" "4:lan:1" "5:wan"
+		;;
 	*)
 		echo "Unsupported hardware. Network interfaces not initialized"
 		;;
@@ -170,6 +175,10 @@ ipq40xx_setup_macs()
 		wan_mac=$(mtd_get_mac_ascii devinfo hw_mac_addr)
 		lan_mac=$(macaddr_add "$wan_mac" 1)
 		;;
+	p2w,r619ac*|)
+		wan_mac=$(cat /sys/class/net/eth0/address)
+		lan_mac=$(macaddr_add "$wan_mac" 1)
+		;;
 	esac
 
 	[ -n "$lan_mac" ] && ucidef_set_interface_macaddr "lan" $lan_mac
diff --git a/target/linux/ipq40xx/base-files/etc/hotplug.d/firmware/11-ath10k-caldata b/target/linux/ipq40xx/base-files/etc/hotplug.d/firmware/11-ath10k-caldata
index c4d9dd78e3..a4740310d2 100644
--- a/target/linux/ipq40xx/base-files/etc/hotplug.d/firmware/11-ath10k-caldata
+++ b/target/linux/ipq40xx/base-files/etc/hotplug.d/firmware/11-ath10k-caldata
@@ -60,6 +60,7 @@ case "$FIRMWARE" in
 	glinet,gl-s1300 |\
 	linksys,ea6350v3 |\
 	mobipromo,cm520-79f |\
+	p2w,r619ac-* |\
 	qcom,ap-dk01.1-c1)
 		caldata_extract "ART" 0x1000 0x2f20
[... 72 lines cut ...]
--- a/target/linux/ipq40xx/patches-5.4/901-arm-boot-add-dts-files.patch
+++ b/target/linux/ipq40xx/patches-5.4/901-arm-boot-add-dts-files.patch
@@ -10,7 +10,7 @@ Signed-off-by: John Crispin <john@phrozen.org>
 
 --- a/arch/arm/boot/dts/Makefile
 +++ b/arch/arm/boot/dts/Makefile
-@@ -837,11 +837,58 @@ dtb-$(CONFIG_ARCH_QCOM) += \
+@@ -837,11 +837,61 @@ dtb-$(CONFIG_ARCH_QCOM) += \
  	qcom-apq8074-dragonboard.dtb \
  	qcom-apq8084-ifc6540.dtb \
  	qcom-apq8084-mtp.dtb \
@@ -55,6 +55,9 @@ Signed-off-by: John Crispin <john@phrozen.org>
 +	qcom-ipq4019-e2600ac-c2.dtb \
 +	qcom-ipq4019-habanero-dvk.dtb \
 +	qcom-ipq4019-pa2200.dtb \
++	qcom-ipq4019-r619ac-32m.dtb \
++	qcom-ipq4019-r619ac-64m.dtb \
++	qcom-ipq4019-r619ac-128m.dtb \
 +	qcom-ipq4019-rtl30vw.dtb \
 +	qcom-ipq4019-u4019-32m.dtb \
 +	qcom-ipq4019-wpj419.dtb \

我根据QSDK公开部分改了下lan wan的配置
```
上面的代码有什么错误的地方吗？

PussAzuki, 2020-12-28:
我记得在op官方开发版内核更新到5.4.83时还是正常的 后面我就在等大变化了

PussAzuki, 2020-12-28:
```diff --git a/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c b/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c
index ed96f38bb7..04f50808c9 100644
--- a/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c
+++ b/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c
@@ -1403,8 +1403,6 @@ netdev_tx_t edma_xmit(struct sk_buff *skb,
 	/* Check and mark VLAN tag offload */
 	if (unlikely(skb_vlan_tag_present(skb)))
 		flags_transmit |= EDMA_VLAN_TX_TAG_INSERT_FLAG;
-	else if (!adapter->edma_cinfo->is_single_phy && adapter->default_vlan_tag)
-		flags_transmit |= EDMA_VLAN_TX_TAG_INSERT_DEFAULT_FLAG;
 
 	/* Check and mark checksum offload */
 	if (likely(skb->ip_summed == CHECKSUM_PARTIAL))

这是我手动打1207时的patch到op主线上 然后复制到x-wrt这边来用git diff 后发现的缺少的东西，属于之前的901和902补丁上的改动

PussAzuki, 2020-12-28:
按照上面的删除掉那两条之后，一切恢复正常，希望能修复你这多出来的bug

ptpt52, 2020-12-29:
并不难确定是这两行的问题，这两行应该是不能删的。

PussAzuki, 2020-12-29:
但你之前的版本(1217之前)都是删了的，之前也确实没遇到任何问题，现在手动删了后我这边也没有任何问题

PussAzuki, 2020-12-29:
你可以用1207的代码来对比1217及以后的代码

ptpt52, 2020-12-29:
这个差异我知道，怎么重现这个问题呢？我重现不出来

PussAzuki, 2020-12-29:
我不确定其他IPv6上有没有此问题，但我在6in4上测试出来了

PussAzuki, 2020-12-29:
最简复现步骤：
1.安装AGH
2.设置一个IPv6 DNS上游(plain/DoT/DoH/...)
3.点击 应用 后点击 测试上游DNS
4.等待片刻，如果提示无法使用则确定是你的没删代码的问题，反之则不是

x-wrt-dev, 2020-12-29:
AGH 是什么？具体怎么安装？

PussAzuki, 2020-12-29:
AGH是adguardhome 
```
opkg update
opkg install adguardhome
```

PussAzuki, 2020-12-29:
stubby也是能测出来的

PussAzuki, 2021-02-27:
请问一下为什么会把一个用了近一千日的补丁给漏掉一点内容呢？

https://github.com/x-wrt/x-wrt/blob/26b2067c7ab6668f6db44a94fb99cd8af700db77/target/linux/ipq40xx/patches-5.4/901-essedma-disable-default-vlan-tagging.patch#L75

我可以证明漏了上面两行操作后编译的固件均 6in4网络环境下 在路由器上使用基于ipv6的DNS服务器均会无法使用 (我这边ipv6原生是nd而不是pd的不好搞，只好6in4将就了事)

ptpt52, 2021-02-27:
@PussAzuki  这样改可以吗
```
diff --git a/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c b/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c
index ed96f38bb7..94db66129b 100644
--- a/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c
+++ b/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/essedma/edma.c
@@ -1403,8 +1403,6 @@ netdev_tx_t edma_xmit(struct sk_buff *skb,
        /* Check and mark VLAN tag offload */
        if (unlikely(skb_vlan_tag_present(skb)))
                flags_transmit |= EDMA_VLAN_TX_TAG_INSERT_FLAG;
-       else if (!adapter->edma_cinfo->is_single_phy && adapter->default_vlan_tag)
-               flags_transmit |= EDMA_VLAN_TX_TAG_INSERT_DEFAULT_FLAG;
 
        /* Check and mark checksum offload */
        if (likely(skb->ip_summed == CHECKSUM_PARTIAL))
```

PussAzuki, 2021-02-27:
对的，就是这两行

closed 2021-03-09 by PussAzuki