x-wrt/x-wrt #153: Bug? Xiaomi Mi Router R3 2.4G wifi not work well

lihexali, 2021-01-16:
**BUG: Xiaomi Mi Router R3 sometime can not  connect 2.G wifi network**
I had a Xiaomi Mi Router R3 flashed latest x-wrt. It works well, but it will got a problem that 2.4G wifi will not work well sundenlly. The mobile phone shows that 2.4G wifi network is connected, but can not contract with the internet or even can't connect the 2.4G wifi.  
** Temporary solution  **
To solve this problem, I had to reboot the router, or reboot the 2.4g wifi radio in Luci manager interface.

** Please tell me why this bug produced, does it is a code bug or a hardwear bug?"

lihexali, 2021-01-18:
没人遇到相同的问题吗？2.4Gwifi网络时不时的连不上，需要重启才能解决

Mr-XeNeS, 2021-08-30:
I have the same problem - WiFi 2.6GHz does not work at all even after factory reboot.

Not working build: x-wrt-9.0-b202108291236-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade

Last version that worked properly: x-wrt-9.0-b202107130301-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade

ptpt52, 2021-09-09:
> I have the same problem - WiFi 2.4GHz does not work at all even after factory reset.
> 
> Not working build: x-wrt-9.0-b202108291236-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade
> 
> Previous version that worked properly: x-wrt-9.0-b202107130301-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade

just bug in this version `b202108291236`
and will be fixed in next version.

devfof, 2023-08-15:
@ptpt52 The same problem is happening to me ( my xiaomi router r3 ) in 2.4 GHz mode, but 5 GHz is working just fine ).
I'm using [ X-WRT 23.06_b202306142256 ( r23543+1-5a1acb7e9e6 )] and [ LuCI Master ( git-23.164.79444-968c21f )].
My router connects and functions properly after restarting, but after 30 to 40 minutes it starts to disconnect again.